ttlappalainen/NMEA2000 #303: Parser when there are multiple sources!

Alyavin, 2022-12-17:
Timo Hi. The question is, I have several gps sources in the Nmea2000 network! When using the ParseN2kGNSS function, how can I choose where from the network I can get this data? I want to receive gps from my chosen source.

ttlappalainen, 2022-12-17:
You need to use tN2kDeviceList class. It is automated module to keep track of devices ont the bus.
- On setup you create device list object pN2kDeviceList = new tN2kDeviceList(&NMEA2000);
- Then you can request tDevice object with pN2kDeviceList in several ways. The strict way is
Device=pN2kDeviceList->FindDeviceByName(Name); where name is exact device NMAE2000 NAME = 64 bit device information.
Since each NMEA2000 device for one manufacturer should have unique unique number you can use
Device=pN2kDeviceList->FindDeviceByIDs(ManufacturerCode, UniqueNumber);
Basically it is possible that if you have GPS and e.g., cabin temperature sensor from same manufacturer, they may have same unique number since their class is different. Using NAME is safest way, but using only ManufacturerCode, UniqueNumber and getting trouble has probalility of winning in lottery.
Note that if you get Device==0, it has not yet been resolved.
- Then get Source=Device->GetSource(); And when you recive Gps message you check that sender is Source.
- To take care of posible address claiming, you check in loop that is ther any changes in sources with pN2kDeviceList->ReadResetIsListUpdated()

so...

```
#include <NMEA2000_CAN.h>
#include "N2kDeviceList.h"
...
tN2kDeviceList *pN2kDeviceList;
uint8_t MainGPSSource=0xff;
...
void setup() {
  ...
  pN2kDeviceList = new tN2kDeviceList(&NMEA2000);
}
...
void HandleGNSS(const tN2kMsg &N2kMsg) {
  if ( N2kMsg.Source==MainGPSSource ) {
    // We got specific GPS data
   ...
}
...
void loop() {
  NMEA2000.ParseMessages();
  if ( pN2kDeviceList->ReadResetIsListUpdated() ) {
    const tNMEA2000::tDevice *pDevice=pN2kDeviceList->FindDeviceByIDs(MfgCodeGa#include <NMEA2000_CAN.h>
#include "N2kDeviceList.h"
...
tN2kDeviceList *pN2kDeviceList;
uint8_t MainGPSSource=0xff;
...
void setup() {
  ...
  pN2kDeviceList = new tN2kDeviceList(&NMEA2000);
}
...
void HandleGNSS(const tN2kMsg &N2kMsg) {
  if ( N2kMsg.Source==MainGPSSource ) {
    // We got specific GPS data
   ...
}
...
void loop() {
  NMEA2000.ParseMessages();
  if ( pN2kDeviceList->ReadResetIsListUpdated() ) {
    const tNMEA2000::tDevice *pDevice=pN2kDeviceList->FindDeviceByIDs(MainGPSMfgCode,MainGPSUniqueNumber);
    if ( pDevice!=0 ) MainGPSSource=pDevice->GetSource();
  }
}rmin,MyGarminUniqueID);
    if ( pDevice!=0 ) MainGPSSource=pDevice->GetSource();
  }
}
```

See also DeviceAnalyzer example. With it you can get mfg codes and unique numbers for your devices or even full NAMEs

Alyavin, 2022-12-17:
thank you very much

ttlappalainen, 2022-12-19:
Most common problem is that you do not call NMEA2000.ParseMessages continuously in loop or there is long delays on loop blocking that call. Try e.g., BatteryMonitor example - it should be shown on DeviceAnalyzer.

Alyavin, 2022-12-19:
Timo when you connect the device with the DeviceAnalyzer program to Lowrance everything that is from the Lowrance device is all displayed, but there is no DeviceAnalyzer device in the list! It does not see itself, but the other is all there

ttlappalainen, 2022-12-20:
Why it should list itself? Either MFD:s does not list itself on their device list.

Alyavin, 2022-12-20:
Timo look, I take with the help of your library from my lowrance elite ti 2 gps coordinates using ParseN2kPGN129025 then I add some value to them and give back to my lowrance elite ti 2 using SetN2kPGN129025(N2kMsg, Latitude, Longitude), but from because I don't see my Source, it turns out mixing my coordinates and coordinates from lowrance elite ti 2. I see the source of coordinates from lowrance elite ti 2 , it has 1. But I don't see my source. How can I make it so that I can also see my source NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,10); I understand it's 10, but I don't see it! I hope you understand me.

ttlappalainen, 2022-12-21:
I do not understand. So you have device reading GPS from Lowrance. You modifu coordinates and send new coordinates to the bus. Which device is then trying to use those coordinates? I think you can not push them to Lowrance, since that uses it's own coordinates internally.

On your device itself you can at anytime request it's own source with NMEA2000.GetN2kSource();

If you do not see your device on Lowrance device list, then you should check that is your device actually sending anything. Your tranceiver may be broken, disabled, misconnected etc.