ttlappalainen/NMEA2000 #305: Usage of SeasmartToN2k ?

dagnall53, 2022-12-29:
With thanks to your earlier help, I have managed to (easily!) use the  N2kToSeasmart conversion,
But I wanted to implement the reverse   "SeasmartToN2k  function. 
I have the 0183 ($PCDIN) message in a buffer, (buf)  and expected that I could just call  the function to result in the N2kMsg type (N2kmsg). But the "time stamp" is confusing me. 
If I call `SeasmartToN2k(buf, millis(), N2kmsg);`  
 I get `"cannot bind non-const lvalue reference of type 'uint32_t&' {aka 'unsigned int&'} to an rvalue of type 'uint32_t' {aka 'unsigned int'}" `

Is there a working example of the code (in this direction) to explore?  The examples all seem to convert N2K > $PCDIN , but not the opposite direction.. 

With the  "other direction", I was (apparently?) able to work with timestamp just being mills()? 
`N2kToSeasmart(N2kMsg, millis(), buf, 500)`

ttlappalainen, 2022-12-29:
uint32_t Timestamp;
tN2kMsg N2kmsg;
SeasmartToN2k(buf, Timestamp, N2kmsg);

dagnall53, 2022-12-29:
YES.. Thanks.. 
I managed it with a very similar construct, just making a 'time' variable, and setting that to millis() before calling the SeasmartToN2k

I presume this sets the N2K message timestamps so that similarly timed messages get treated together. 
    ```
uint32_t time;
  time=millis(); 
SeasmartToN2k(buf, time, N2kmsg);
```

Many thanks
Dagnall

dagnall53, 2022-12-29:
PS I am trying to add "0183 things" to the NMEA0183Handlers.cpp.  
I will send you the Cpp when I have added some more. 
This one was just added as "HandleDIN"

closed 2022-12-29 by dagnall53