ttlappalainen/NMEA2000 #306: Several fluid sensors with one ESP32

Pepinaco, 2022-12-31:
Hello Timo, I want to use several fluid sensors with one ESP32. Is it possible to send data from the sensor to different PGN? for example 12505,12506,12507. In which file I have to do the modification.
Thank you in advance.

ttlappalainen, 2022-12-31:
PGN 127505 is for fluid level. 127506 is for DC detailed status and 127507 for charger status.

If you send several fluid levels from single device, you just use different fluid instance value for each sensor and also set fluid type.

You can use TemperatureMonitor as base and modify settings to match fluid level device. Then you write example:
```
// *****************************************************************************
void SendN2kFluidLevels() {
  tN2kMsg N2kMsg;

  if ( FluidScheduler.IsTime() ) {
    FluidScheduler.UpdateNextTime();
    SetN2kFluidLevel(N2kMsg, 0, N2kft_Water, WaterTank1Level/WaterTank1Capacity*100, WaterTank1Capacity);
    NMEA2000.SendMsg(N2kMsg);
    SetN2kFluidLevel(N2kMsg, 1, N2kft_Water, WaterTank2Level/WaterTank2Capacity*100, WaterTank2Capacity);
    NMEA2000.SendMsg(N2kMsg);
    SetN2kFluidLevel(N2kMsg, 2, N2kft_Fuel, FuelTank1Level/FuelTank1Capacity*100, FuelTank1Capacity);
    NMEA2000.SendMsg(N2kMsg);
  }
}
```

greenhouse69, 2023-01-09:
I understood that a combination between instance and the fluidtype would make it different. So, is this correct?:

....
    FluidScheduler.UpdateNextTime();
    SetN2kFluidLevel(N2kMsg, **0, N2kft_Water**, WaterTank1Level/WaterTank1Capacity*100, WaterTank1Capacity);
    NMEA2000.SendMsg(N2kMsg);
    SetN2kFluidLevel(N2kMsg, **0, N2kft_Fuel,** FuelTank1Level/FuelTank1Capacity*100, FuelTank1Capacity);
    NMEA2000.SendMsg(N2kMsg);
    SetN2kFluidLevel(N2kMsg, **1, N2kft_Water**, WaterTank2Level/WaterTank2Capacity*100, WaterTank2Capacity);
    NMEA2000.SendMsg(N2kMsg);
  }
}

ttlappalainen, 2023-01-10:
You have to use different instance for each tank.
```
FluidScheduler.UpdateNextTime();
SetN2kFluidLevel(N2kMsg, 0, N2kft_Water, WaterTank1Level/WaterTank1Capacity*100, WaterTank1Capacity);
NMEA2000.SendMsg(N2kMsg);
SetN2kFluidLevel(N2kMsg, 1, N2kft_Fuel, FuelTank1Level/FuelTank1Capacity*100, FuelTank1Capacity);
NMEA2000.SendMsg(N2kMsg);
SetN2kFluidLevel(N2kMsg, 2, N2kft_Water, WaterTank2Level/WaterTank2Capacity*100, WaterTank2Capacity);
NMEA2000.SendMsg(N2kMsg);

```

greenhouse69, 2023-01-10:
Thank you timo, It's a real pleasure to have you here. So in the same device, the maximum number of tanks is 16 (0-15) regardless of type of fluid. And another question, Is it possible for two devices to have the same Instance simultaneously on network?

ttlappalainen, 2023-01-10:
It should be. Devices should handle same instances from different devices. Unfortuntely this does not seem to work with all MFD:s. I had problems with my Garmin GMI 20, which did not handle that. It required each tank having different instance even it game from other device.

greenhouse69, 2023-01-10:
I will be testing on a Garmin xs742xs and a lowrance elite 7 ti some setups with multiple devices and Instances. Thanks for the help, I'll report the results.

greenhouse69, 2024-10-22:
I have a very basic question, PGN 127505 sends Tank instance. Tank instance values are 0-15 or 0-254? I have this confusion because I think I remember that the maximum values are 16 or maybe I have seen some code that has confused me. If you can clarify it I would appreciate it. Thanks

ttlappalainen, 2024-10-23:
PGN 127505 tank instance can be 0-15 not 16. It has only 4 bit for instance. Do not mix to e.g., temperature instances, which has 8 bit value 0-252. There is reservation for one reserved, error and NA.

greenhouse69, 2024-10-23:
I meant that the maximum number of values is 16 : from 0 to 15 (4 bits = 1nibble= 2 exp4 = 16 different values. Thanks