ttlappalainen/NMEA2000 #307: Device stops broadcasting when it is the only sender on the bus

GerhardMalan, 2023-01-01:
Hello Timo,

Thank you for your fantastic work on this library!

I am building a few devices for our boat's (https://svsunnyspells.com/) NMEA/SIMNET bus. It's a bit tricky to test things as we are currently in Tasmania/Australia and the boat is on the hardstand in the Aegean :-(

During testing of CANBUS communication I noticed that a simple sending device (ESP32 with sn65hvd230 module) stops transmitting
after a while if there are no other devices to acknowledge the messages. At least I think this is what is happening from my limited research. If I have (only) two devices connected on the bus, the sender has to be reset after the receiver is started, otherwise no messages are ever received. For example, I am working on code for the receiving device. Whenever I flash the code to the receiving device, I have to follow up with a reset of the transmitting device. I have been able to confirm with oscilloscope that it is the sender that is not transmitting.

Is this normal behaviour? Also, I'm concerned that if devices on the NMEA2K bus do not start in the correct sequence, a transmitting device may start first, and then stop because it never received an acknowledgement. Is there a way to periodically reset a transmitting device if it receives no acknowledgement? No one else seems to have come across this issue, so maybe I am doing something else wrong?

Thank you again for your work on this library and ongoing support - it must have been an enormous task.

Gerhard

agk1190, 2023-01-02:
Gerhard,

I noticed this exact same problem a month or so ago while working on my own project. 

In my scenario, I had only my two PCBs (ESP32 with MCP2562) hooked up to the NMEA2000 bus. One PCB (main) was running firmware based off of the DataDisplay2 example in this library. The other PCB was running the MessageSender example. 

Anytime I would reprogram the main PCB or it would crash, the MessageSender PCB would stop sending messages until I reset it.

I did not have enough time to investigate further since this was for a college project and it did not affect the "real world" use. 

I too am curious as to why this is happening.

-Alex

ttlappalainen, 2023-01-02:
This is feature of CAN. In CAN protocol there is ACK bit. If that is set some other device must acknowledge reception. If there is no acknowledge, controller will try to resend frame. Acknowledge will happen under controller level. I tried some time ago find a way not to set ACK bit, but did not quicly found solution and had to give up, since I had other work to do.

I have used Teensy and ESP32 and they have continued sending, when it is possible. The only problem is that library has internal buffer, which will be then sent out in one burst. The goal is to catch that sending error earlier and not to buffer anything in that case. Library should then also automatically restart address claiming when bus is running again. Anyway for most likely I will do modifications only for Teensy and ESP32 drivers first.

ttlappalainen, 2023-01-02:
And by the way for testing I prefer to have one device programmed with example ActisenseListenerSender and use my NMEA Simulator. Then you can fully test your device. The best is naturally buy a commercial box so you will get full bus analyzing features with it. See https://acrossoceansystems.com/index.php/product/nmea2000-nmea0183-simulator/

GerhardMalan, 2023-01-02:
Hi Timo,

Thank you for your quick response! 

That is good to know that I am not the only one to experience the problem.  The safest thing to do in this instance is then probably to place a delay in the boot of all sending devices to ensure at least one listener is active on the bus after power up. For the moment this will be enough for me. I will not be able to test all of this until May/June when we go back to the boat in Greece.

I have one other question about mode setting as well. When setting mode `N2km_ListenAndNode` during setup, the interrupts for incoming messages (irq 1) are not raised in the interrupt handler, so no messages can be received. Sending messages still works as expected and other IRQs are still raised. I fixed this by setting the mode in the MEA2000 OnOpen callback. Is this expected behaviour? Would you like me to start a different GitHub issue for that?

Gerhard

ttlappalainen, 2023-01-02:
Must be something else. N2km_ListenAndNode  should have been changed to default mode and I use it as default on all my projects. The only difference to N2km_NodeOnly mode is that in N2km_NodeOnly mode library passes all automatic message forwarding.

GerhardMalan, 2023-01-04:
Ok thanks Timo and thank you again for your support.