ttlappalainen/NMEA2000 #309: Force a tN2kSyncScheduler event to happen immediately.

preeve9534, 2023-01-18:
I have a project that uses PGN 127501 to report binary status of multiple channels.
The requirement is to generate periodic (i.e. scheduled) status reports and also to
report immediately on a state change on any channel.

Periodic reporting is obviously satisfied by the scheduler, but at the moment it seems
like I have to step outside the scheduler function to implement the immediate, ad-hoc,
state change reporting.  This requirement for two side by side mechanisms seems a
little inelegant.

If there was a mechanism for forcing a tN2kSyncScheduler to immediately execute
its 'next' event and then simply resume its scheduling, then it seems to me that
would be a nice solution.

I wonder if you have any thoughts?

P

ttlappalainen, 2023-01-19:
That would require extra variable to tN2kSyncScheduler. It could be OK for PC, if you like to waste memory, but not with these small devices.

I do not either see any need for that feature. Handler callbacks has been done on your call to ParseMessages, so you can easily add respond immediately to the queue. In my solutions I simply call specified message sender. Note that it is not good to start to solve message data on sender, since that may delay sending and cause variation in message timing. In my devices I have other task handling reading of data. Then message senders has data immediately available and does not spend time for reading data. With this method in my devices message timing variates mostly less than 1 ms. I have seen that on some devices (even on big manufacturers) variation is several tenths ms. Many also does have cumulative error on scheduling, which is caused if you do like NextTime=millis()+Period;

Normally, when you get request, you should respond immediately.
- If you will be requested with either ISO request or request group function (without offset or period change), you just call you message sender.
- If you get change commanded by switch bank control message or command group function, you handle change and respond immediately new status or ack for command group function. I have not seen clear definition for command group function should you respond also with commanded PGN. I think should be OK also respond immediately.

In above cases you do not touch scheduler.

If you get request with request group function with period or offset change, definion is not well done. So I do next:
- If requested offset is NA or 0, I respond immediately and do not change offset.
- If there is other offset, I just set that and respond also with message. In this case if offset change is to e.g., 1 ms forward, there will be two messages very near of each other.
- If there is period change, I just setup new period. Respond as above.
- In any case some parameter is not accepted and it is not broadcast, there will be nack response.

Above satisfies certification tests.

ttlappalainen, 2023-01-19:
Actually noticed that it should be enough to have function ForceNextTime(FromNow:uint64_t); Then it would enable next schedule to given time, but with UpdateNextTime next scheduled time would be calculated.

preeve9534, 2023-01-19:
Thanks for taking the time to make this clarification. Timo.

I pretty much have the situation you describe at the moment - the thing I occasionally see is two messages very close together: the scheduled "heartbeat" ticking away and then the exceptional response overlayed on this pattern. It doesn't cause an operational issue per-se, but in some sense is a bit of a waste of bus bandwidth.

Interesting that you had the idea of a ForceNextTime() function.

My homebrew scheduler just includes a force() function which triggers the next scheduled event immediately and then restarts the schedule immediately.  There is no real impact on resource utilisation. However, my scheduler relies on millis() and therefore exhibits rollover issues. 

From my perspective the benefit is that the transmission path for a particular PGN is pipelined into a single process model based around the scheduler which (I think) makes the code cleaner and easier to conceptualise. And eliminates the multiple redundant transmit issues.

If you think it useful, I can take a look at adding a force function. Let me know.

ttlappalainen, 2023-01-19:
Currently I do not need ForceNextTime, but if you think it is useful, just look at it. It should not effect anything.

tN2kScheduler is just copy of my own library tmsScheduler. I had similar tusScheduler to work with us resolution.

closed 2023-01-20 by preeve9534