ttlappalainen/NMEA2000 #310: tN2kSyncScheduler instantiation problem

preeve9534, 2023-01-19:
I have some firmware which supports setting the transmit period and offset of a PGN.
I persist the set values in EEPROM and need to assign these to a scheduler when the
firmware starts.

The documentation for tN2kSyncScheduler more or less anticipates this scenario:
```
* For proper synchronization you define tN2kSyncScheduler for each of your 
 * messages as disabled (default constructor) at beginning. On library setup
 * define OnOpen call back with tNMEA2000::SetOnOpen. When OnOpen will be
 * called, it initializes each message scheduler with period and offset.
 ```
But the TemperatureSensor example doesn't follow this recipe: it simply supplies
defined values in the class constructor.

Following the instructions rather than the example I create a default, disabled,
scheduler with:
```
tN2kSyncScheduler MessageScheduler();
```
And then...
```
void onN2kOpen() {
    MessageScheduler.SetPeriodAndOffset(mySavedPeriod, mySavedOffset);
}
```
Which, I believe, should satisfy my needs. However, compile fails with:
```
In file included from src/SIM108.cpp:306:0:
src/definitions.h: In function 'void onN2kOpen()':
src/definitions.h:37:22: error: request for member 'SetPeriodAndOffset' in 'MessageScheduler', which is of non-class type 'tN2kSyncScheduler()'
```
Which leads me to think that the default class constructor failed.

If I replace the default (i.e. empty) constructor call with a call containing some arguments (as in the
TemperatureSensor example), say
```
tN2kSyncScheduler PGN127501Scheduler(false, 2000, 500);
```
then everything compiles as expected.

So, the default constructor for tN2kSyncScheduler seems to be broken in some way.
I note that the three constructor parameters are assigned defaults in the class definition
in the library, so I am not quite sure why use of the empty constructor is failing.

Obviously I have a work-around (just supply some dummy arguments to the constructor),
but this still leaves a mis-match between the instructions in the class comments, the
example supplied in TemperatureSensor, and what happens on the ground.

ttlappalainen, 2023-01-20:
Ahh, this is common error. If you do want to use default constructor without arguments, you have to use:
`tN2kSyncScheduler MessageScheduler;`
With any amount of arguments you can use:
`tN2kSyncScheduler MessageScheduler(false);`

preeve9534, 2023-01-20:
Thanks Timo, This is something I should have known, Sorry for being a muppet.
P

ttlappalainen, 2023-01-20:
Well, I did the same mistake and tested it first with () - took a moment to remember that it is wrong. During writing code normally, I never try () for default constructor.

closed 2023-01-20 by preeve9534