ttlappalainen/NMEA2000 #311: broken links

greenhouse69, 2023-01-25:
Hello everyone,
I have been able to review the documentation at (https://ttlappalainen.github.io/NMEA2000/index.html). I have been able to verify that all the links with nmea.org are broken:
List of all Manufacturer Codes
NMEA2000 Standard.
List of NMEA 2000 registered devices.
List of NMEA 2000 registered companies.
Device class and function codes.
ISO address claim.
Group function PGN 126208 handling.
Heartbeat PGN 126993.
Thanks for your great work.

ttlappalainen, 2023-01-25:
Can not do anything. NMEA orginazation has changed their structure and maybe hide those documents to be used only by members. Or they may be open on other path, which has not yet been found by Google

greenhouse69, 2023-01-25:
Correct Timo, the nmea.org website has been updated and modernized. It was simply to warn that these links are broken in the documentation.

ttlappalainen, 2023-01-31:
So there is e.g., https://web.archive.org/web/20190531120557/https://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
Can that be used as reference or will it disappear some time?