ttlappalainen/NMEA2000 #312: PGN130316-SetTemperature

greenhouse69, 2023-02-08:
I have been able to read the documentation and I do not understand the concept of this value: set temperature. I have seen it in PGN130316 and in PGN130312. If I use PGN130312 -> SetN2kTemperature(N2kMsg, 0, 255, N2kts_SeaTemperature, CToKelvin(Temperature), N2kDoubleNA); I get Field 5: Set Temperature= Data not aviable. ( this is normal) but if I use PGN130316 ->(SetN2kTemperatureExt(N2kMsg, 0, 255, N2kts_SeaTemperature,CToKelvin(Temperature), N2kDoubleNA); I get Field 5: Set Temperature= 3276.7 Kelvin ( this is something I didn't expect) It exists a reason for this? What is the use of Set Temperature? I'm confused. Thank you.

ttlappalainen, 2023-02-08:
There was error in 130316 parser and setter. I wonder how I or any other have not noticed it before. Temperature monitor example does not use it and Message sender example uses set value so error will not appear.

It is fixed now in just published version.

greenhouse69, 2023-02-08:
Timo, as always fast and efficient. I don't know how to thank you for your great dedication. Thank you and keep investigating

greenhouse69, 2023-02-08:
Confirmed, all right now. What function does the Set Temperature have? Thanks again.

greenhouse69, 2023-02-08:
Thanks Timo, that's what I supposed, but it's better to confirm so as not to reach wrong conclusions. Thanks again.

greenhouse69, 2023-02-16:
Hi everyone, can someone confirm if more than one PGN 130311 message can be sent from the same device? With PGN 130312 and PGN130316 I have no problem sending several temperatures to the bus if the temperature instances are different but when I try it with PGN 130311 in the NMEA Reader it only shows a PGN line showing alternately the pairs of temperatures and temperature instance.

ttlappalainen, 2023-02-17:
There is no problem to sent it, but unfortunately depends of receiver does it separate temperatures with with different field 2 code. E.g., Actisense reader does not separate messages with different instances. E.g., my NMEA Simulator does. 130311 is anyway deprecated and should not be used.

greenhouse69, 2023-02-17:
Hi everyone, can someone confirm if more than one PGN 130311 message can be sent from the same device? With PGN 130312 and PGN130316 I have no problem sending several temperatures to the bus if the temperature instances are different but when I try it with PGN 130311 in the NMEA Reader it only shows a PGN line showing alternately the pairs of temperatures and temperature instance.
Thanks, so you recommend neither using 130311 and 130312 (both obsolete), or just 130311. Greetings.

ttlappalainen, 2023-02-17:
If you are making it only for your boat, just use newest your listening device supports in order 130316, 130312, 130311. There is no need to send other than one. 
If you make commercial, you must have configurable system so that user can select PGN according his requirements.

greenhouse69, 2023-02-18:
Ok, very thanks