ttlappalainen/NMEA2000 #314: Windlass PGN 128777

mar0997, 2023-02-19:
Hi Timo - 

Thank you for all of the help.  This library and examples have helped me to set up a number of things on my boat.  However, I have been stumped by trying to add the windlass PGN 128777.  I have set up a very simple output, but when I read this in my Actisensce NMEA reader it does not show all fields.  I only see field 1 even though I do see that there are 8 fields in the length.  I know I am probably doing something wrong, but I have set this up similar to the other PGNs that I am outputting.  Can you please offer me some advice on what I am doing wrong?  Thanks!

![Windlass Output Code](
[image]
https://user-images.githubusercontent.com/90991895/219933861-c81e98e8-8bba-4f4a-a7df-3e2d74ea3347.jpg)

ttlappalainen, 2023-02-19:
PGN is so new that Actisense reader does not support it and so it does not show any information. My NMEA Simulator has support, but free version shows data parsed for first 100 messages. So if you arrange so that you do not have anything else on the bus sending data, you can catch message parsed. Commersial version parses messages and also allows 3M messages recorded at time.

mar0997, 2023-02-19:
Thank you for the quick reply.  I’ll try it on my Garmin MFD to see if it works there.  Supposedly Garmin supports this PGN so this should be a simple test.  Again I appreciate your help.

Tarakona2850, 2023-02-19:
Garmin, only supports nmea183 .  MFDs only partially. the GMI 20, on the other hand, in its entirety. there it is also possible to reset the chain length from the display, as well as to set the size of the chain drum

virtuvas, 2023-02-26:
@Tarakona2850 : GMI10 and GMI20 support windlass via custom NMEA0183 sentences simulating AutoAnchor black box and NOT via PGN128777. 
FWIW, if anyone wants them I've got them sorted and working via one of my teensies and a reed switch.