ttlappalainen/NMEA2000 #318: two independent devices connected to the nmea 2000 network

greenhouse69, 2023-04-23:
I currently have two independent devices connected to the nmea 2000 network.  The problem is that when I list the devices the plotter (lowrance) only finds one, but they both work on the plotter, it is as if one is taking over both devices.  This is my code in the TEMP device:
TEMP Product:

void setup() {
.....
/
    NMEA2000.SetProductInformation("DBLD5Q2CD8", 
        100, // 
        "TEMP",  
        "1.0.0.1 (2023-01-01)",  
        "23001001", 
        0xffff, // load equivalency - use default
        0xffff, // NMEA 2000 version - use default
        0xff // Certification level - use default 
    );
    
    NMEA2000.SetDeviceInformation(23001001, 
        130, 
        75,
        2046, 
        4    
   );

Read  NVS last NodeAddress, by default is 30,

And this is the code in the Level device:
LEVEL Product:

void setup() {
.....
/
    NMEA2000.SetProductInformation("DBLD5Q2CD8", 
        100, // 
        "LEVEL",  
        "1.0.0.1 (2023-01-01)",  
        "23001001", 
        0xffff, // load equivalency - use default
        0xffff, // NMEA 2000 version - use default
        0xff // Certification level - use default 
    );
    
    NMEA2000.SetDeviceInformation(23001001, 
        150, 
        75,
        2046, 
        4    
   );

Read  NVS last NodeAddress, by default is 30,

Both have the default nodeaddress set to 30, I have tried changing the default nodeaddresses but no change. 

Any hints? what can I be doing wrong? is this normal? the plotter recognises the data from both devices, but I would like to understand how it works. Thank you.

greenhouse69, 2023-04-24:
I was able to solve it by reducing the resolution of the Dallas Temperature sensor, the default resolution is 12 -> 0.0625°C to 9 -> 0.5°C: 
sensors.begin(); 
sensors.setResolution(9);
I think the problem could be that the resolution is so precise that it slows down the loop too much (750ms) and the nodeaddress cannot be updated.  
I hope it can be useful to someone in the same situation as me, be careful with the default resolution of the Dallas Temperature.  We keep on studying.

svdrummer, 2023-04-24:
i am not sure how to do it in arduino land, in machine code on commercial products, we send the "get temperature" command. Go away and do things then com back and read the temperature. On the dallas site, it had the command number to use.
Hope this helps. I wasted so much time on this when i first started.

ttlappalainen, 2023-04-24:
Some things:
- NMEA NAME will become from SetDeviceInformation, which must be unique for all devices on same bus. Even different function/class codes defines uniques NAME, I'll prefere to use different unique ID for each different device.
- Unique ID max is 2097151. This is documented on my document. Setting ID bigger that that just runs 21 bit ID over and does not show the one you expected on MFD
- I prefer to start each device with different source. Also prefer to follow possible source changes with ReadResetDeviceInformationChanged. If source has been changes, save it some how (EEPROM) and restart next time with saved source.
- Using default Dallas temperatory reading library is doomed. As document says you should do ParseMessages in loop faster than 10 ms cycle. Dallas library may stop reading for 700 ms, which totally destroys NMEA2000 handling. You have to yse asynchronous and non blocking reading for Dallas sensors.

greenhouse69, 2023-04-24:
Thanks Timo for the advice, I'm still testing.