ttlappalainen/NMEA2000 #320: using PGN 126208 for Binary switch control?

u1550l, 2023-04-26:
Hello Timo,
this is a great library!

From issue https://github.com/ttlappalainen/NMEA2000/issues/128 I understand that today the recommended way to control binary switches is using Pgn 126208 (and then probably do a write group function).
In your code I found the group function handler, but I did not find a predefined function to send a 126208 message (i.e. a SetN2kPGN126208(..) function

Did I miss it?

My application: 
building a set of display and output modules for use in my camper van environment.
Marine chartplotters are not suitable here. So, I need to build both user interface and hardware interface.

Thanks
Richard

u1550l, 2023-04-27:
hello all,
I think, this might be interesting to you:
I asked ChatGPT about PGN 126208. See my attached pdf.
[chatGPT-on-Pgn126208.pdf](https://github.com/ttlappalainen/NMEA2000/files/11343431/chatGPT-on-Pgn126208.pdf)

ttlappalainen, 2023-04-28:
Have you read the library document? There is link to proper document [Group function PGN 126208 handling](https://web.archive.org/web/20170609033039/http://www.nmea.org/Assets/20140109%20nmea-2000-corrigendum-tc201401031%20pgn%20126208.pdf)

Attached is handler for binary group function. It is from old project and do not remember does it work completely. I use in my own projects general handler, which can handle any PGN. Unfortunately I can not publish it.

[N2kGroupFunctionBinaryStatus.zip](https://github.com/ttlappalainen/NMEA2000/files/11349193/N2kGroupFunctionBinaryStatus.zip)

u1550l, 2023-05-03:
Thanks Timo,
for the link to the group function document on web.archive
and for your example code.

u1550l, 2024-03-03:
Hello Timo,
it's been a while since I started my n2k project. unfortunately I had to put it on hold since then.

Now, I thought, I could get progress with the example you provided here: https://github.com/ttlappalainen/NMEA2000/issues/320#issuecomment-1526870529
But, I did not find a function to set up the 126208 send frame.
So, this is what I tried right now:

```
#define PRIORITY_DO_NOT_CHANGE    0x08
#define RESERVED_4_BITS_SET_TO_1  0x0f
#define COMMANDED_PGN             127501L              // Binary Status Report 127501

void SetN2kPGN126208cmd(tN2kMsg &N2kMsg, unsigned char DestinationId,
                     unsigned char FieldNoOfParam, unsigned char FieldValue) {
    N2kMsg.SetPGN(126208L);
    N2kMsg.Priority    = 3;
    N2kMsg.Destination = DestinationId;
    N2kMsg.AddByte(N2kgfc_Command);     // field 1
    N2kMsg.Add3ByteInt(COMMANDED_PGN);  // field 2 : Commanded PGN
                                        // field 3 : Priority Setting
                                        // field 4 : NMEA Reserved
    N2kMsg.AddByte(PRIORITY_DO_NOT_CHANGE | RESERVED_4_BITS_SET_TO_1<<4);
    N2kMsg.AddByte(1);                  // field 5 : Number of Pairs .. to follow
    N2kMsg.AddByte(FieldNoOfParam);     // field 6 : Field No of commanded param
    N2kMsg.AddByte(FieldValue);         // field 7 : Value of commanded param
}

void sendSwitchCommand(unsigned char DestinationId, unsigned char FieldNoOfParam, unsigned char FieldValue)
{
    tN2kMsg N2kMsg;
    SetN2kPGN126208cmd(N2kMsg, DestinationId, FieldNoOfParam, FieldValue);
    NMEA2000.SendMsg(N2kMsg);
}
```

My problem:   The Actisense NMEA reader shows unexpected fields in the frame:

[image]

What am I doing wrong?

Thanks for your help!
Richard

ttlappalainen, 2024-03-04:
Group Function PGN depends of PGN it is handling, so it is not easy to build common handler. I also prefer to change your SetN2kPGN126208cmd name to e.g., SetN2kGroupFunctionCommand127501.

In you command you do it right. Seems that Actisense Reader shows it wrong. I tested you data with my NMEA Simulator under Tools-NMEA2000 Commands...-Any Message , where you can build binary code and send it to bus.
PGN:126208, 01 0d f2 01 f8 01 03 01
Shows on NMEA Simulator
[image]
but on Actisense Reader
[image]

NMEA Simulator is far more better NMEA2000 tool than Actisense Reader, but unfortunately parser works only on paid version.

u1550l, 2024-03-04:
Thanks a lot for your immediate answer.
Your suggestion for a good function name is exactly what I was looking for. 

Unfortunately the NMEA Simulator is currently out of range for my hobby project. (It would cost 650,- euro). But good to know. May be it's an option for the future.

Now that I know that the message is correct, I will start writing the handler function.

ttlappalainen, 2024-03-04:
If you are controlling switches, some devices still supports old 127502, where you call it once for switch, which state you want to change. Others should be set NA.

Some has made own Simulator hw from example ActisenseListenerSender. Then you can bye just sw license by contacting me directly with email.