ttlappalainen/NMEA2000 #321: Battery Status Voltage Signed or Unsigned ?

ieb, 2023-04-28:
https://github.com/ttlappalainen/NMEA2000/blob/1a959925d4686b5d709b654b9ddb319d3e0ca2f8/src/N2kMessages.cpp#L768

Canboat says the battery voltage in 127508 is an  uint16_t with 0.01, see https://canboat.github.io/canboat/canboat.html
Won't matter for anything < 33V, but for 42V Lithium packs its might as the -ve bit is set.

Sorry if this is a duplicate.

ieb, 2023-05-02:
my mistake I must have miscounted the digits, sorry about that. No issue in the library.

Is the documentation at CanBoat incorrect stating that the field is a 2ByteUDouble ?

ttlappalainen, 2023-05-02:
Yes. As I wrote PGN 127508 battery voltage is 2 byte int with 0.01V resolution.

ieb, 2023-05-07:
opened in error, so closing.

closed 2023-05-07 by ieb

rszemeti, 2024-11-15:
Just a note:  This is incompatible with Canboat.js and thus with SignalK   ... Your lib uses a signed uint for the voltage field, canboat.js uses a unsigned uint, the result is -ve voltages come out as 654 volts etc in SignalK when sent fom your lib.   I don't know definitively which one is correct!

https://github.com/canboat/canboat/blob/9914fdb2595e6f548aa831957e2338f8de48067a/analyzer/pgns.json#L16665

ttlappalainen, 2024-11-15:
127508 battery voltage field is defined as signed with range +/- 327.64 V

rszemeti, 2024-11-15:
Noted, I shall see if I can get the Canboat people to change theirs then,  I will report back.