ttlappalainen/NMEA2000 #331: Two esp32 (m5atom) devices see each other but not others

mgrouch, 2023-05-29:
I’m using your library on esp32 with m5atom.
Somehow
two esp32 (m5atom) devices see each other but not others. 
And vice versa 2 other devices (non m5atom based) see each other’s messages but
not m5atom ones. I’m lost at this point. 

thanks!

ttlappalainen, 2023-05-30:
Either there is CANH/CANL mixed or for some reason m5atom boards has wrong baud rate. Easiest is to swap CANH/CANL wires on CAIS3050G. If that does not help modify NMEA2000_esp32 speed setting on constructor: speed(CAN_SPEED_250KBPS).

CAIS3050G is just tranceiver and isolation module and should no have way to change CAN clock.

mgrouch, 2023-05-30:
I just swapped H and L on both m5 atoms and everything started working!
Jeez. I spent two days trying to isolate it in software. Thinking it’s software issue. 
My NMEA cable when I cut it came with unconventional coloring. So I connected blue in whatever was marked blue in m5atom. 

Thanks for getting back and solving it so quickly. !

closed 2023-05-30 by mgrouch

ttlappalainen, 2023-05-30:
These boxes are traditionally made for CAN and so they do not follow N2k colouring.

mgrouch, 2023-05-30:
So by swapping wires I effectively can have 2 somewhat independent logical buses on one physical CAN bus? Are there bad side effects?
Just curious…

Thanks

ttlappalainen, 2023-05-30:
No. It will cause errors for both. It works unpredictable and gets worse with more traffic.

mgrouch, 2023-05-30:
Ok. Thanks