ttlappalainen/NMEA2000 #332: Custom PCB for Teensy + MCP2562 for NMEA2000 to PC (Actisense NGT-1 alternative)

cheeseheap, 2023-06-02:
Hi Timo,

First of all: thank you so much for the work you've done here!

My goal is to connect our NMEA2000 (Raymarine SeaTalkNG) network to our onboard PC, which is running OpenCPN. I noticed you can buy an [NGT-1 from Actisense](https://actisense.com/products/ngt-1-nmea-2000-to-pc-interface/) to do this, however I'm not comfortable with the outrageous price. I noticed some others have tried the same so I've been playing around and bought a Teensy 4.0, MCP2562, some other parts and a soldering iron. I haven't tested the contraption that I created yet, however I already know that I want to create something more "elegant" for the long term.

That's why I've investigated creating a custom PCB for this, and this seems easier than I thought! Please find attached a PDF, which is the easiest way to get an idea of what I'm trying to do (zoom in). Also attached a 3d visual and a Gerber file I want to use to order the part. FYI I created this on https://easyeda.com/, which seems very intuitive to use, even for a newbie like me! The goal was to create a simple PCB that can be put together by a non-expert with limited skills like me. For instance, I am only using THT. I am planning to "place" this on top of the Teensy by using some tall header pins. On the right-top of the PDF you'll see a hole for above the Teensy pin Vin (5V). Below that GND, 3.3V etc. Note that I tried to make the board work for both Teensy 3.2 and 4.0 by for instance routing the Tx to pin 3 (for Teensy 3.2) and to pin 22 (for 4.0).

I have two questions for you (or anyone else viewing this):

1. Do you think the PCB looks good? I understand that I myself am responsible for what I order, I'd just like to have a second pair of eyes take a look at it. I have used [this schematic](https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Images/teensy32_uniso.png) as an example. Is that a good starting point? Note I am planning to add the capacitor and 4.7K resistor directly into the MCP2562 holes.
2. For the code I am planning to use your [ArduinoGateway.ino](https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/ArduinoGateway/ArduinoGateway.ino). Am I correct in saying that I can use that code, but for Teensy I just need to update to "#define N2k_CAN_INT_PIN 21" to pin 23?

FYI if there is anyone here who would like this PCB as well, I am required to order more than one. Send me a message and if you live in Europe and are interested. For cost of board + shipping I am more than happy to send to you!

Thanks so much!

Cheers,

Cees (NL)

[PCB_MCP2562 CAN hat_2023-06-02.pdf](https://github.com/ttlappalainen/NMEA2000/files/11638633/PCB_MCP2562.CAN.hat_2023-06-02.pdf)
[Gerber_MCP2562 CAN hat_2023-06-02.zip](https://github.com/ttlappalainen/NMEA2000/files/11638634/Gerber_MCP2562.CAN.hat_2023-06-02.zip)
[image]

ttlappalainen, 2023-06-03:
1) PCB looks good, but I prefer to use isolated tranceivers for connecting bus to PC. Schema "Teensy 3.2, simple unisolated" you refer is only for devices, which are totally isolated on other side. Check isolation schema examples like "Teensy 3.2, simple isolated" and add protection as on other picture. Since you are using it from PC, you do not need to power Teensy with other DC-DC converter and cut USB power from it.

2) You do not need to modify #define N2k_CAN_INT_PIN 21 since that is only for external MCP2515 CAN controller. Teensies has internal CAN controller.

cheeseheap, 2023-06-03:
Hi Timo,

Thanks! I will try to make a second version using the ISO1050.

Couple of followup questions:

1. Regarding the ISO 1050: I noticed the image shows a second DC power supply on the left bottom. Do you think I can leave this out and power it just from the Teensy? (via PC USB, the PC is a desktop so not dependent on battery)
2. I want to stick to THT as much as possible as it seems easier. The problem is I can't find the ISO1050 in THT version. Is it "doable" to bend the pins on an [SMT component like this](https://www.reichelt.nl/nl/nl/ge-soleerde-can-zendontvanger-3-0-5-5v-sop-8-iso-1050-dubr-p188896.html?PROVID=2809&gclid=CjwKCAjwyeujBhA5EiwA5WD7_eqlj7VzluCvmPWGGzq0HnC-eKqBlOo3U4ax5PnuPDkhPhTLyocM7hoC7lwQAvD_BwE) and use it as THT?
3. I noticed the ISO1050 comes in "DUB" and "DUBR". Do you have any idea what the difference is?

Thanks again!

ttlappalainen, 2023-06-03:
1. Teensy side is best to power from Teensy 3.3V out. Teensy 3.2 inputs are 5V tolerant, but Teensy 3.5-4.1 are not.
2. Why you should use THT? Just draw board to use SMT. Pins may not be long enough for THT.
3. DUB is tube and DUBR is reel packed. You can find that information by going to Texas Instrument side.

cheeseheap, 2023-06-03:
Thanks so much! Apologies I didn't see your comment regarding the DC-DC converter earlier.

I watched some tutorials on SMT and am going to give it a try. Attached the updated PCB with ISO1050, this one was much easier to make. Would be super helpful if you could take another look!

Last two questions (hopefully):

1. Should I add a capacitor on the 3.3V side or do you think that is not necessary?
2. We turn the PC off by cutting power to it. Therefore the USB (and the Teensy) will not have power. Can this cause issues if half of this "system" (the ISO1050 transceiver, powered by BUS) is powered and the other half (the controller in Teensy) is not?

[image]
[Gerber_ISO1050 CAN hat_2023-06-03.zip](https://github.com/ttlappalainen/NMEA2000/files/11642614/Gerber_ISO1050.CAN.hat_2023-06-03.zip)
[PCB_ISO1050 CAN hat_2023-06-03.pdf](https://github.com/ttlappalainen/NMEA2000/files/11642615/PCB_ISO1050.CAN.hat_2023-06-03.pdf)

ttlappalainen, 2023-06-04:
I can not say anything from the picture except that nice 3D. It does no tell anything about schema.

1. Yes as in picture "More protection for tranceiver". I also prefer to add protection diodes and L2 on N2k side.
2. Should not have problem. You should study datasheets of parts you are using. You can find that information on datasheet under "8.3.3.3 Undervoltage Lockout and Fail-Safe". On data sheets there is also description about how boar layout should be planned. You need to have clear separation between isolated and unisolated side. I can not say from picture does that meet proper separation, but looks that you are crossing wires so that it does not meet requirements.

cheeseheap, 2023-06-04:
Thanks! I've been doing some more research after your comment and learning a lot :) I'm assuming when you talk about diodes / L2 you are talking about those seen on [this schematic](https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/src/images/iso_trans.png)? I've been looking at those parts (specifically the diode) and I'm worried I don't have the skill / tools to solder that small. Therefore I will order two boards, one with and one without the diode. Regarding the L2, I'm assuming this is a Common Mode Choke? If not, what do you mean by L2? If you have a part number that would be great, as it's difficult for me to figure out exactly what I need.

Apologies if I'm asking obvious questions! I'm a beginner and just getting started. FYI please see my progress below.

Board without Diodes
[image]

With diodes
[image]

ttlappalainen, 2023-06-05:
You do not need two boards with and without diodes, since you can just leave diodes not connected.

Yes, L2 is common shoke filter. I'll try to find code... It is also small tiny part, but you can do the same as with diodes and put jumpers.

Your board probably works, but it does not follow isolation rules. Your left side wires runs under the chip and there are left side through holes on right side. There should be clear separation between left and right side of ISO1050 chip as in document https://www.ti.com/lit/ds/symlink/iso1050.pdf 11.2 Layout Example.

N2k cables normally also has shield around. You can use 4 pin connector and leave shield hanging around to cause risk of short or use 5-pin connector where to connect shield. On board shield pin will be left unconnected. Note also that as in my examples I used 5-pin connector with pin order as in connectors use with DeviceNet, where NMEA2000 has been inherited. You can also use 3.5 mm pin separation to save space.

cheeseheap, 2023-06-08:
Hi Timo. I've been making some progress based on your feedback, thanks again! I am working with a PCB supplier and also partially dependent on the parts that they can supply. Also, I have shifted my strategy a bit and thought it would be cool to be able to create something that you can just "push" a Teensy 4.0 with pins into. Please see below, let me know your thoughts!

Also, any recommendations on that choke filter? I am hoping I will be able to source it from the supplier that I am getting the PCB from (similar for the diode, they are able to provide me that part).

[image]
[image]

ttlappalainen, 2023-06-12:
Board looks better. 

I am not specialist with hw design. R-78e5.0-0.5 is anyway DC-DC converter and will cause some noise to CAN V+if there is no filtering. 

One possibility for filter is SRF2012-121YA.

cheeseheap, 2023-06-14:
Thanks! I've made some final updates to the board and added a common mode filter to it. I can't get my hands on the part that you mentioned, but I found [this WCM2012F2SF-121 as an alternative](https://datasheet.lcsc.com/lcsc/2009171440_TAI-TECH-WCM2012F2SF-121T04_C357308.pdf). Let me know if you think that would be OK. The basic parameters are:

Impedance: 120Ohm ±25%
Frequency: 100MHz
DC resistance max: 0.30 Ohm
Rated Current: 400mA
Rated Voltage max: 50Vdc
Withstand Voltage max: 125Vdc
IR: 10M Ohm

Final board layout below. Will order soon. Could you let me know if there is anything that you think is wrong? For instance, does it matter which way the common mode filter is connected? (I couldn't find this in the documentation)

Thanks a ton! I'll post my findings after I receive the board and hook it up to the NMEA 2000 bus on board (btw, I created a prototype with the MCP2562 and it works well!)

[image]
[image]

ttlappalainen, 2023-06-15:
- What is the point of P5-P7?
- And how about U11 connector pin order? Why you are crossing CAN Tx/Rx. With different pin order you could avoid vias.
- Why that is name U11? U is normally mark for chips.
- Left side 100nF capacitor could be more near of the chip. It is preferd to keep bybass capacitor as near as possible and traces as short as possible.
- About Common mode filter you can compare their charasteristics.

cheeseheap, 2023-06-20:
HI Timo. Thanks for your feedback (again)! You made me think and I have re-done the design again. I'm going to order this tomorrow so let me know if you absolutely think this will not work. Note I am not aiming for perfection here, just something that will (likely) work. Also note: forget about P5-P7 and U11 (this was automatically created by the software). I'm thinking of a different way of mounting the Teensy to this. I hope to be able to show you with pictures of the final result.

A few things that I have tried to do with this change:
- Prevent Rx/Tx and CanH/CanL from crossing (thanks for the tip)
- Capacitors as close to chip as possible (thanks again for tip)
- When I do need to cross signal lines, make sure its the GND (I read somewhere this is better).
- Additional text on PCB to explain everything.

Lmk if you find anything off!

[image]
[image]

cheeseheap, 2023-07-13:
Hi folks,

IT WORKS! :) Last update here for those that may be interested in doing this in the future. PCB Gerber file attached (obviously I can't guarantee anything as I am not a professional in PCB design). Tag me in any questions you may have!

I ordered the PCB from China using [easyeda.com](https://easyeda.com/) and subsequently ordering through their parent company JLCPCB.com. I used that provider because they have an easy UI and they were able to (very cheaply) supply some of the parts that I couldn't find easily here. They happened to have a deal where they would solder for a small fee the parts that they were supplying. This made things VERY easy for me. I was especially worried about having to solder the choke (its tiny). See pictures below. When it arrived I only needed to solder the capacitors, DC-DC converter (left over from my prototype) and and the male/female pin headers. I was then able to slide it onto the Teensy no problem.

You'll notice I changed the design compared to the previous post. I changed the way the PCB is oriented compared to the Teensy and this allowed me to (at least to a degree) respect the ISO 1050 isolation rules. It's effectively a "CAN HAT" for the Teensy 4.0. I'm sure that from a best-practices POV there is still a whole lot wrong with this design, but hey it works. Last thing for me to do is to create a permanent mounting solution. I was thinking of modifying [this bracket](https://www.pjrc.com/panel-mount-free-standing-brackets/) for that.

Overall the cost amounted to about €100 in parts total. A whole lot less than an Actisence NGT-1!

Parts list:
- Teensy 4.0 €40
- PCB's (had to order 5) including ISO1050, choke, diode and terminal block (incl shipping) $43
- DC-DC converter €9
- Some parts I had lying around (capacitors, male/female pin headers etc.)
- Box to mount it in

[image]
[image]
[image]
[image]
[image]
[image]
[Gerber_ISO1050 CAN hat V2.1_2023-07-13.zip](https://github.com/ttlappalainen/NMEA2000/files/12036510/Gerber_ISO1050.CAN.hat.V2.1_2023-07-13.zip)

cheeseheap, 2024-05-25:
Update: we’re almost a year on and the pcb is still working great!