ttlappalainen/NMEA2000 #334: Platformio package does not install dependencies automatically

alexg-k, 2023-06-22:
First of all, thanks for this awesome library!

Now, adding
```
lib_deps = 
ttlappalainen/NMEA2000-library@^4.18.7
````
to my projects `platformio.ini` results in compilation errors. I have to manually add
```
lib_deps = 
        coryjfowler/mcp_can@^1.5.0
	ttlappalainen/NMEA2000_mcp@^1.1.2
	ttlappalainen/NMEA2000-library@^4.18.7
````
to resolve the issue. Can you please add these dependencies to [here](https://registry.platformio.org/libraries/ttlappalainen/NMEA2000_mcp/insights/dependencies), so the dependencies will be installed automatically?

I still have various issues integrating your lib in my platformio project, but I have to investigate those further.

ttlappalainen, 2023-06-23:
You should use mcp_can under my git, not others, if they has not been validated.

alexg-k, 2023-06-23:
I am unable to find a `mcp_can` repo under your username on github or on the platformio registry.

ttlappalainen, 2023-06-23:
https://github.com/ttlappalainen/CAN_BUS_Shield

alexg-k, 2023-06-23:
I see, thx! The repo has a platformio library file, but is not listed on the platformio registry under your username. Could you register it over there?

alexg-k, 2023-06-23:
PR #335 fixes the integration of this library in a platformio project.

Please Note: I am not sure, that this dependency stack is true for every platformio project using this library.

ttlappalainen, 2023-06-23:
I cancelled your PR. It will not work as you did it.

alexg-k, 2023-06-23:
I understand. In case somebody else wants to use your library with PlatformIO and my hardware setup (Arduino MKR Wi-Fi 1010 & MKR Can Shield), I ended up with these dependencies: 
````
lib_deps = 
	;ttlappalainen/CAN_BUS_Shield@^1.20 ; <-maybe in the future?
	https://github.com/ttlappalainen/CAN_BUS_Shield.git
	ttlappalainen/NMEA2000_mcp@^1.1.2
	ttlappalainen/NMEA2000-library@^4.18.7
````

closed 2023-06-23 by alexg-k

ttlappalainen, 2023-06-24:
Why do you need those lib_deps. I have PlatformIO projects for different MCU:s and never used lib_deps. You can simply copy NMEA2000, CAN_BUS_Shield and NMEA2000_mcp under you project lib. Or what is even better you copy CAN_BUS_Shield and NMEA2000_mcp under lib_MKR folder and add that folder as library in platformio.ini. Then you can have own definition for e.g.. ESP32 for same project and use lib_ESP.

Be aware that if you modify library files, you need to redo modification for every library update.