ttlappalainen/NMEA2000 #336: Can Bus 2 on Teensy 4.0

MrMount11, 2023-07-02:
Hi Timo,

If I want to use the second can bus on a teensy 4.0 (pins 0 & 1) is this correct?

#define NMEA2000_TEENSY_CAN_BUS 2

Thanks, Rich

ttlappalainen, 2023-07-03:
Please read the documents.

For Teensy 4.0 you need NMEA2000_Teensyxx library. For that you select bus 2 by defining:
#define NMEA2000_TEENSYX_CAN_BUS tNMEA2000_Teensyx::CAN2
You find this on file NMEA2000_CAN.h or document https://ttlappalainen.github.io/NMEA2000/_n_m_e_a2000___c_a_n_8h.html

MrMount11, 2023-07-05:
Thanks Timo,  I had read the documentation but missed one critical piece,
which was to define the bus before including the library.  Once  I
rearranged that, it worked beautifully. Kind of obvious in hindsight!

Rich