ttlappalainen/NMEA2000 #338: PGN 126983

RemcoHalman, 2023-07-20:
Hello, 

First of all thanks for all te effort in producing this amazing library.

I'm looking for a way to integrate with PGN126983 (alert messaging). What the purpose it it would be to filter this out of the NMEA stream and record the alarms on a SD card. The idea is to implement this and use it for charter/rental of the boat.

Could you help with a initial route for this? It's a fast package but thats the only reference i found on the library. I would really appreciate it!

ttlappalainen, 2023-07-21:
Do you have devices supporting/sending 126983?

As any bus message it can be also catched. First you need data description for message, which is a bit complex for alert messages.

RemcoHalman, 2023-07-21:
I have a few EmpirBus units to use for pushing the alarms. And a Garmin GPSMAP 8612 to use and display the alerts. 

What would be a good reference? Because if we use the listener in readable format is it a massive amount of info with just a few devices.

ttlappalainen, 2023-07-21:
Not sure did I understood your needs. Anyway example DataDisplay simply parses only FluidLevel and WaterDepth. With similar method you can handle only 126983 and do with it what ever you want. There is no parser for 126983 so that you should write by yourself.

RemcoHalman, 2023-07-21:
Sorry if i was unclear on the need. i'm going to use this library in a prototype for an Alarm logging. 

The purpose is to register the N2K alert with the system time PNG timestamp on a SD card. Later on want to add 4/5G support to push it to a cloud service. Sort of Fleet management.

I'll look at the DataDisplay then. My experience with C++ is limited so can time consuming thing. If I have a base could you check if i'm on the correct path?

RemcoHalman, 2023-07-25:
Ok this is to hard for me to start to be honest. For now I'll work with the 127501 binary indicators to know the alarms. 

Perhaps with time ill be able to get it running.

Do you know if there is a reliable maintained python library just like you cpp one? Perhaps i can start from there.

For now the issue can be seen as #closed
Thank you for the quick responses and initial help!

ttlappalainen, 2023-07-25:
There may be some limited libraries. Before my library, there was no free full N2k libraries with certification readiness.

RemcoHalman, 2023-07-25:
Thats what my thoughts were. Thank you for the answer! 

Ill try to create the project with my current knowledge. The project should be scheduled Q1 2024 for commercial use

ttlappalainen, 2023-07-25:
Or you can order project from me. In that case contact me directly with email.