ttlappalainen/NMEA2000 #339: Question - Device Power

boatarmor, 2023-07-31:
Timo,

Could you please advise if the power supply configuration below is acceptable?  (Board will be Teensy 4.0)

Many Thanks,

Jeremy Hudson

[image]

ttlappalainen, 2023-08-01:
On designing hw I prefer to always read carefully each chip datasheets. For VCC2 prefered supply voltage is 4.75 - 5.25
[image]

ttlappalainen, 2023-08-10:
As beginner you are very rigth path. Some notes anyway.
- One very BIG NO! Do not use delay(xxx) anywhere in loop of function called from loop. This destroys idea of tasks. Library has been designed so that it can be used on single core non tasking system so NMEA2000.ParseMessages(); behaves as task and it should be called in loop as fast as possible. Also you have to take care that not other "task" in loop will not stop loop for long time (max. <20 ms randomly, normally <1 ms). If e.g., reading some sensor takes long, you have to create "task" for it so that you start reading and then by timer or polling ready state (normal with ADC). In general use asynchronous or non blocking methods.
- For sync scheduler constructor is tN2kSyncScheduler(bool Enable=false, uint32_t _Period=0, uint32_t _Offset=0). Now you have different period for all. NMEA 2000 defines default pressure period for 130314 2000 ms so I prefer to use that. You can find default values on beginning of NMEA2000.cpp where PGN:s has been listed.
- I think you do not need to use 3 devices. Multidevice is very rarely needed. Just give different instance for each pressure. 
- On sending SetN2kPressure second parameter is SID. Either use N2kUInt8NA or increase value after every send. Its idea is to tell that value has been cathed at same time even related messages may have been sent at different time. In your case I would just use N2kUInt8NA.
- As mentioned about "tasking" you have now in loop 2 "tasks". I separate data reading to different task since they are in a way independent of data sending. Also in that way it is easier to handle cases, when your data needs special handling for making it non blocking. If there is e.g., slow ADC with conversion time 200 ms, I have reader task, which starts conversion and scheduler for 210 ms from start. Then task just check scheduler and when it is time it also checks conversion ready. If conversion is not ready, I e.g., add 10 ms for sheduler. The reason for using scheduler and not just ready state is that in most cases checking scheduler is much more faster than checking ready state. This may be with ADC but every different senso system may need to design own "tasking" to make it fast and non blocking.
- With own value reading task you would then handle Serial.println(...) with scheduler. Now you are printing it fast all the time and wasting resources.
- About layout. You have rather big indention. Normally 2 or 4 spaces will be used. Also block closing will be in with same indention as start. So. e.g.,
  as I like
  if (...) {
  }
  or 
  if (...)
  {
  }

boatarmor, 2023-08-11:
I can't say thank you enough for your response.  Sincerely appreciate the time you took to provide direction & offer advice.  Understood on each bullet and will make revisions accordingly - and will try my best to break the habit of big indentions!  

I've got all hardware ordered and should arrive sometime next week.  Will report back with results and final code to be shared.

Thank you again and talk soon.

Best,

Jeremy H.