ttlappalainen/NMEA2000 #340: Stops receiving on some networks depending on SetN2kCANReceiveFrameBufSize

waimate01, 2023-08-13:
Hello Timo,

I have put off asking for months because I'm pretty sure it's nothing to do with your fine code, and everything to do with me. But I am stymied :)

I have an ESP32 with a MCP2551 driver. I have a small Raymarine network on my desk -- a course computer and an MFD. I run your sample programs and they work great! I can see all messages flowing around, and I can transmit. Beautiful !

Then I take my device to an actual vessel, which has Raymarine, same model course computer, and a larger number of MFDs.

When running in N2km_ListenOnly mode, all works fine and I see lots of messages all the time.

But when running in N2km_ListenAndNode, I see a burst of messages for a brief period of time and then I never hear another message. I am also unable to send. The device is not dead - it responds in other ways - but the NMEA2000 traffic seems locked up.

The interesting thing, I think, is that by calling SetN2kCANReceiveFrameBufSize with a large number, the time until the problem occurs can be increased - up to 20 seconds or so. I have turned on frame level debug, and I see some messages about a lost frame and some orphan frames just before the problem occurs, but maybe these are just symptoms and not causes?

I keep coming back to the fact it all works fine on the minimal network on my desk. The problem only occurs near water :) The network on the boat is otherwise functioning reliably and I have checked my termination resistor situation many times, so I'm reasonably sure I'm ok there.

I would be very grateful if you could please give me some clues on where I have gone wrong, because it sure beats me :)

With thanks

-- Ian
    Sydney, Australia

ttlappalainen, 2023-08-13:
MCP2551 is not good choice with ESP32. It requires 5V power and then also outputs 5V and ESP32 is not 5V tolerant. So have you installed also some voltage level converter. As I have mentioned in many documents I prefer to use either MCP2562 or ISO1050 (or some other isolated tranceiver) when isolation is required. See https://ttlappalainen.github.io/NMEA2000/page_h_w_set_up.html

I use ESP32 a lot and it is also used commercially. I have also found that some old ESP32 chip CAN controller get stuck in case lots of errors in the bus. I have fix for that but have not published it yet. So one reason could be that due to MCP2551 you have errors in the bus and that stuck ESP. You should get rare frame errors during debug. If you do get them often, there is something wrong in your bus.

waimate01, 2023-08-20:
Thank you for your response. Turns out it's nothing to do with SetN2kCANReceiveFrameBufSize, it just looked that way.  Upon first attempt to transmit, it gets CAN_IRQ_ERR, CAN_IRQ_ERR_PASSIVE, CAN_IRQ_ARB_LOST and CAN_IRQ_BUS_ERR. It appeared to still be functioning because of queued inbound messages.  Odd because it works fine on a smaller network.  Oh well, on to other things :)

ttlappalainen, 2023-08-20:
As I mentioned this happens, but normally rarely. You have some problem with your network.

Did you check your MCP2551 powering - it does not work with 3.3V? It requires 5V Vcc, which causes problem that ESP32 IO is not 5V tolerant.