ttlappalainen/NMEA2000 #341: SetN2kPGN129285 Causes Infinite Recursion on Build

brian-r-calder, 2023-08-16:
On macOS using cmake 3.26.0-rc6, clang 14.0.3, macOS Ventura 13.4.1(c), attempts to build the library for local use (with simple makefiles) fail due to an infinite recursion by the inline version of SetN2kPGN129285 in N2kMessages.h:

inline void SetN2kPGN129285(tN2kMsg &N2kMsg, uint16_t Start, uint16_t Database, uint16_t Route,
                        bool NavDirection, bool SupplementaryData, char* RouteName)
{
   tN2kNavigationDirection NavDirection1 = NavDirection?N2kdir_reverse:N2kdir_forward;
	SetN2kPGN129285(N2kMsg, Start, Database, Route, NavDirection1, (uint8_t)SupplementaryData, RouteName);
}

There is a static version of SetN2kPGN129285() in N2kMessages.cpp, but the compiler has no way to distinguish between them and attempts to keep unrolling the inlined version until the stack runs out.

Since there are no dependencies on this call in the remainder of the library, commenting the inline out allows the system to build, but another solution would be preferred.

ttlappalainen, 2023-08-16:
Definetely inline should be:
```
inline void SetN2kPGN129285(tN2kMsg &N2kMsg, uint16_t Start, uint16_t Database, uint16_t Route,
                        bool NavDirection, bool SupplementaryData, char* RouteName)
{
   tN2kNavigationDirection NavDirection1 = NavDirection?N2kdir_reverse:N2kdir_forward;
   tN2kGenericStatusPair N2kSupplementaryData = SupplementaryData?N2kDD002_Yes:N2kDD002_No;
	SetN2kPGN129285(N2kMsg, Start, Database, Route, NavDirection1, RouteName, N2kSupplementaryData);
}                        

```

brian-r-calder, 2023-08-25:
OK, would you like me to convert this to a pull request?

ttlappalainen, 2023-08-28:
I fixed that to my code and will update it soon with other fixes/improvements.