ttlappalainen/NMEA2000 #342: seasmart.cpp function SeasmartToN2k() warnings on checksum, byte, source, pgnLow and pgnHigh may be used uninitialized

virtuvas, 2023-08-18:
hi Timo,

wanted to report that long ago, but always forget. Nothing serious, just annoying, I'm getting the following warning messages when compiling with the NMEA2000 libraries of the last few months under IDE2.1.1 (not PlatformIO haven'd done the switch yet).
Below is using the latest library NMEA2000@4.18.7: 

```
c:\Users\Vassilis\OneDrive\V_DoC\Programming\arduino\libraries\NMEA2000-master\src\Seasmart.cpp: In function 'bool SeasmartToN2k(const char*, uint32_t&, tN2kMsg&)':
c:\Users\Vassilis\OneDrive\V_DoC\Programming\arduino\libraries\NMEA2000-master\src\Seasmart.cpp:182:3: warning: 'checksum' may be used uninitialized in this function [-Wmaybe-uninitialized]
  182 |   if (checksum != nmea_compute_checksum(buffer)) {
      |   ^~
c:\Users\Vassilis\OneDrive\V_DoC\Programming\arduino\libraries\NMEA2000-master\src\Seasmart.cpp:171:17: warning: 'byte' may be used uninitialized in this function [-Wmaybe-uninitialized]
  171 |     msg.Data[i] = byte;
      |     ~~~~~~~~~~~~^~~~~~
c:\Users\Vassilis\OneDrive\V_DoC\Programming\arduino\libraries\NMEA2000-master\src\Seasmart.cpp:150:14: warning: 'source' may be used uninitialized in this function [-Wmaybe-uninitialized]
  150 |   msg.Source = source;
      |   ~~~~~~~~~~~^~~~~~~~
c:\Users\Vassilis\OneDrive\V_DoC\Programming\arduino\libraries\NMEA2000-master\src\Seasmart.cpp:139:29: warning: 'pgnLow' may be used uninitialized in this function [-Wmaybe-uninitialized]
  139 |   msg.PGN = (pgnHigh << 16) + pgnLow;
      |             ~~~~~~~~~~~~~~~~^~~~~~~~
c:\Users\Vassilis\OneDrive\V_DoC\Programming\arduino\libraries\NMEA2000-master\src\Seasmart.cpp:139:22: warning: 'pgnHigh' may be used uninitialized in this function [-Wmaybe-uninitialized]
  139 |   msg.PGN = (pgnHigh << 16) + pgnLow;
      |             ~~~~~~~~~^~~~~~
```
Obvs wont affect what I'm doing, I'd rather not get red coloured code on the compile window as it scares me every time 😁 

cheers

V.

ttlappalainen, 2023-08-19:
Will it work properly if you add new line after line 102 static bool readNHexByte(const char *s, unsigned int n, uint32_t &value)
value=(uint32_t)(-1);

virtuvas, 2023-08-19:
yes, will compile with no warnings with this extra line!
thank you.

mind, I'm not using seasmart features at all, so don't know if this line breaks anything (doubt it)

cheers

V.

closed 2023-08-19 by virtuvas