ttlappalainen/NMEA2000 #343: Question: espBerry?

wjquigs11, 2023-09-06:
Has anyone used any of the NMEA2000 libraries with an espBerry? (ESP32 development board with RPI-compatible GPIO header that allows use of HATs) like this:
https://espberry.com/espberry-with-waveshare-dual-isolated-can-bus-interface/
If so, which library works?

ttlappalainen, 2023-09-06:
Not sure has anybody tested, but in principle it should work. You need to use mcp library https://github.com/ttlappalainen/NMEA2000_mcp.

Also if you "automatic" library selector, you need to force mcp library to be used and set proper pins:
#define MCP_CAN_RX_BUFFER_SIZE 50
#define N2k_SPI_CS_PIN GPIO_NUM_16
#define N2k_CAN_INT_PIN GPIO_NUM_17
#define USE_N2K_CAN 1
#include <NMEA2000_CAN.h>

Or you can define it manually:
#include <NMEA2000_mcp.h>

#define MCP_CAN_RX_BUFFER_SIZE 50
#define N2k_SPI_CS_PIN GPIO_NUM_16
#define N2k_CAN_INT_PIN GPIO_NUM_17
#define MCP_CAN_RX_BUFFER_SIZE 

tNMEA2000 &NMEA20001=*(new tNMEA2000_mcp(N2k_SPI_CS_PIN,MCP_16MHz,N2k_CAN_INT_PIN,MCP_CAN_RX_BUFFER_SIZE));

NMEA2000_mcp library does not currently support having two instances, so you can not use both CAN busses on hat.

wjquigs11, 2023-09-07:
Not having two instances poses an interesting problem. I need two busses because my Garmin wind display is not able to change its n2k source. So I need to read wind data from the instrument on can0, modify it based on mast rotation, and transmit it on can1. However, I can do the parsing/processing all on can0, so I probably do not actually need two instances of the library. The question is, how do I get the parsed/process Wind PGN onto can1? Is there an ESP32 way of doing 'parse_wind can0 | cansend can1'? Maybe do SetForwardStream(Serial), read from Serial, and write to can1?

ttlappalainen, 2023-09-08:
I wonder why Garmin Wind display can not change its source. Normally with CAN source means bus source address 0-252. Address claiming (changing source) is very basic requirement for any device connected to bus, so I expect Garmin would not publish device behaving badly. Or do you mean that you can not configure wind display, which wind source it should use?

If you have two CAN bus, you need two library instance to handle them. Also you mention "modify it based on mast rotation". Do you have rotating mast and several wind instruments on the mast from which you must select actual one?

As summary I understood that you have several wind instruments and Garmin display and you can not configure Garmin display, which wind instrument it should use. For this I find two solution:
1. Connect all to one bus and turn off other except active instrument by cutting power line or separating both CAN bus wires.
2. You must have two CAN bus, where you have on can0 all wind instruments and can1 others. In that case you can use some of the MCP2515 can interface modules and for other you use ESP32 internal CAN controller and just ISO1050 CAN tranceiver. Then you use tNMEA2000_mcp for e.g., wind CAN tNMEA2000_esp32 for others. You can then have simply:

```
#define ESP32_CAN_TX_PIN GPIO_NUM_16
#define ESP32_CAN_RX_PIN GPIO_NUM_17
#include <NMEA2000_CAN.h>

#define MCP_CAN_RX_BUFFER_SIZE 50
#define N2k_SPI_CS_PIN GPIO_NUM_34
#define N2k_CAN_INT_PIN GPIO_NUM_35
#define MCP_CAN_RX_BUFFER_SIZE
#include <NMEA2000_mcp.h>

tNMEA2000 &NMEA2000Wind=*(new tNMEA2000_mcp(N2k_SPI_CS_PIN,MCP_16MHz,N2k_CAN_INT_PIN,MCP_CAN_RX_BUFFER_SIZE));

...
void setup() {
  ...
  NMEA2000Wind.SetMsgHandler(HandleNMEA2000WindMsg);
}
```

On wind bus message handler you simply do
```
void HandleNMEA2000WindMsg(const tN2kMsg &N2kMsg) {
  switch ( N2kMsg.PGN ) {
    case 130306UL:
      if (    UsePortWindWane && N2kMsg.Source==PortWindWaneSource 
           || UseStbdWindWane && N2kMsg.Source==StbdWindWaneSource ) {
        NMEA2000.SendMsg(N2kMsg);
      }
      break;
  }
}
```

wjquigs11, 2023-09-08:
I *think* I might be back where I started...
"The PICAN-M HAT utilizes the MCP2515 CAN Controller, and it is controlled per SPI (Serial Peripheral Interface), and we are using the Standard ESP32 and RaspberryPi GPIO signals for the espBerry."
Which to me indicates I need to use the nmea2000_esp32 library, is this correct?
However, a (very crude) workaround might be to read from can0 parse the data using your library, then simply dump the raw output onto can1. So even if I'm restricted to one instance of the library, I can still use both CAN ports.

ttlappalainen, 2023-09-09:
You must use NMEA2000 library on the main NMEA2000 bus. If you  want to use both CAN ports, you need to listen raw CAN data on wind side, parse that and then use library for sendin on display side. You can not send just wild wind PGN to the main bus, since devices does not accept it, if sender is not introduced itself (ISO address claim, product information). Also it is important that your sender device takes care of ISO address claiming to avoid source collision. That is what my library does in background. If your Garmin dislapy accepts only your Garmin wind vane, you must fool for diplay that it esist on the main bus. This you can try to do by reading wind vane device, product and configuration information and setting them to library instance. Then it shows up for display as your wind device. Unfortunately if display is really locked to use only Garmin devices, there may be some proprietary messages between them and they may be hard to handle.

wjquigs11, 2023-09-11:
Thank you, this is great information. 
I'm using https://github.com/buhhe/VolvoPenta-N2K_Interface as a template, but reading raw data from the CAN bus requires some parsing, and you're already written quite a lot of code to do that, so if you know of any examples of using nmea_mcp I would love to see them. I'm confused on the relationship between N2k_SPI_CS_PIN and N2k_CAN_INT_PIN in your code, and MOSI/MISO/SCLK/CS on the ESP32, but I am still researching and learning.

closed 2024-01-21 by ttlappalainen