ttlappalainen/NMEA2000 #351: PGN127488 : return OK for RPM but return 0 for EngineBoostPressure and EngineTiltTrim

gpenot, 2023-09-23:
I try to read PGN127488 with exemple below. The value of EngineSpeed is OK but the value of EngineBoostPressure and EngineTiltTrim is always 0. Have you try to simulate the PGN127488 or have you an idea why the 2nd and 3nd parameter don't work? 
best regards
PS Excuse for my english!
Gilles

Code Of the Listener (ESP32 WROOM)

_double EngineSpeed,EngineBoostPressure;
  u_int8_t EngineTiltTrim;
void HandlePGN127488(const tN2kMsg &N2kMsg) { 
  unsigned char EngineInstance;
  ParseN2kPGN127488(N2kMsg, EngineInstance, EngineSpeed, EngineBoostPressure,EngineTiltTrim);
   Serial.println (EngineSpeed);
  Serial.println (EngineBoostPressure);
  Serial.println (EngineTiltTrim); 
 }__
Code of the Writer
_double EngineSpeed,EngineBoostPressure;
  int8_t EngineTrim;
 void SendN2kEnvironnement(void) {
        EngineSpeed = 3500
        EngineBoostPressure=4.2;
         EngineTrim=1;
        SetN2kPGN127488(N2kMsg, 0, EngineSpeed, EngineBoostPressure, EngineTrim);
NMEA2000.SendMsg(N2kMsg);
}_  
_void loop() {
  SendN2kEnvironnement();
  NMEA2000.ParseMessages();_
}_

ttlappalainen, 2023-09-23:
Did you read the parser description from document? Nearly all values arein SI units so EngineBoostPressure will be provided in Pa (Pascal). Since resolution is 100 Pa, 4.2 will be 0. EngineTrim should work and is in %.

gpenot, 2023-09-23:
I'm a dane! I had read well for the temperature but not for the pressure. Everything is OK now, thank you very much. For information the goal is to instrument my sailboat engine for display on instruments GARMIN and FURUNO. It already works for temperature and engine speed

closed 2023-10-07 by gpenot