ttlappalainen/NMEA2000 #352: References to millis() in N2kDeviceList breaks ESP-IDF

nicklasb, 2023-09-30:
Hi, 
For some reason, these instances of millis() has started breaking ESP-IDF. 
If I change them to N2kMillis() it starts working. 
https://github.com/ttlappalainen/NMEA2000/blob/e90c387b99a69d32d4e1d071f02b4dc33254bdfe/src/N2kDeviceList.h#L625

It is all kind of strange as I have not had any problems with this before, might be some compiler or ESP-IDF-change. 
Either way, it seems more correct to use N2kMillis, right?

Kind regards, Nicklas

ttlappalainen, 2023-10-02:
N2kMillis is OK. It is just a wrapper for different environments and if you look deeper to code both uses esp_timer_get_time() for ESP32.

Strange behaviour anyway.

nicklasb, 2023-10-02:
Ok, will add a PR then.

closed 2024-01-21 by ttlappalainen