ttlappalainen/NMEA2000 #355: Need Explanation on Sendmsg() and ParseMessage()

gpenot, 2023-10-07:
Hello and thank you again for your help. I am currently working on a temperature sensor with a very low power ESP32. To do this, I'm going to put the ESP32 in deep hibernation mode, but I'd like to make sure I understand the logical architecture of the library. 
The SetN2kPGN130311 function only prepares the message. MEA2000. Does Sendmsg() actually send the message on the bus and then I can hibernate sur ESP32 until I need a new acquisition? But in this case, what is the purpose of the call to NMEA2000. ParseMessages() in the loop(); 

Best regards

ttlappalainen, 2023-10-07:
It is not good idea to make device going to sleep all the time. Displays will stop showing data, if it is not sent continuously according PGN defined period. E.g., 130311 period is 500 ms. Note alos that it is deprecated and you shoul use 130316, 130313, 130314.

You can not hibernate device after SendMsg(). It will be buffered to CAN controller and may take some ms before it goes out. Also if you hibernate device, it can not respond to other devices requests and will probably cause that they does not even care of your device. 

As simple conclusion, forget hibernating any NMEA2000 device.

gpenot, 2023-10-09:
Thank,

OK I will not continue on my way.

I use 130311 but it was an example, I use others pgn too. I Don t know
which one is use on Garmin and furuno installed 10 years ago!
Where can I find definition of pgn period as 500ms for 130311?

Have a nice day

Gilles

ttlappalainen, 2023-10-09:
I have listed most PGNs on https://github.com/ttlappalainen/NMEA2000/blob/master/src/NMEA2000.cpp. E.g., 130311 on line 296