ttlappalainen/NMEA2000 #356: ESP32 Guru Meditation Error - ParseMessages

RemcoHalman, 2023-10-16:
Hi All! 

Again thank you for all the hard work!

i'm building a Logger project for the network. All works fine except that after every 10K lines the ESP reboots. 

I ran a stacktrace of the message and it states that the ParseMessages is calling the `tNMEA2000::open()` but crashing on that. While if i look on the N2K device listing of a Garmin MFD there is no issue with the connection to the network. 

FYI 
```bash
Backtrace: 0x400e0c1f:0x3ffcb360 0x400e1be0:0x3ffcb3a0 0x400d3630:0x3ffcb3e0 0x400e6175:0x3ffcb400

  #0  0x400e0c1f:0x3ffcb360 in tNMEA2000::Open() at .pio/libdeps/esp32doit-devkit-v1/NMEA2000-library/src/NMEA2000.cpp:1206
  #1  0x400e1be0:0x3ffcb3a0 in tNMEA2000::ParseMessages() at .pio/libdeps/esp32doit-devkit-v1/NMEA2000-library/src/NMEA2000.cpp:2537
  #2  0x400d3630:0x3ffcb3e0 in loop() at src/main.cpp:69
  #3  0x400e6175:0x3ffcb400 in loopTask(void*) at C:/Users/remco/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

Could you shine a light on the `NMEA2000.ParseMessages()` function? What does it do and is it the requirement to keep the "heartbeat" running?

Thank you in advance!

RemcoHalman, 2023-10-16:
My apologies tried everything but it was a buffering error on my part.

closed 2023-10-16 by RemcoHalman

ttlappalainen, 2023-10-16:
tNMEA2000::ParseMessages() does a lot and it has to be called <=1 ms period. There can be occasional longer delays, but main idea is to call it without delays. On multitasking system one normally need to use small yeild (<= 1 ms) to give time for other tasks.

tNMEA2000::ParseMessages() collects incoming frames and builds messages from them. It responds to some default system requests or commands, runs heartbeat. It also calls defined message callbacks as related message arrives.

On beginning tNMEA2000::ParseMessages() calls tNMEA2000::Open(), which does nothing after first successfull open. So if you have already catched 10k messages, open call has no meaning.

Library is used in several commercial devices also running WiFi at same time. I have not got any reports library causing Guru meditation error.

Have you checked you have enough memory and you use proper data locking between tasks? You can not e.g., call tNMEA2000::SendMsg() from other task without locking task running tNMEA2000::ParseMessages().

RemcoHalman, 2023-10-19:
I had enough memory, only the writing to a buffer and writing somewhere locked out other tasks like ParseMessages()

Thank you for the clarification