ttlappalainen/NMEA2000 #358: Gateway node full emulation.

FarFarSouth, 2023-10-20:
Hi Timo,

As a newbie to your NMEA2000 library, I first want to thank you for sharing your excellent work.
My request is simple: I’m trying to emulate an Actisense NGT-1. I used the ActisenseListenerSender.ino example successfully but noticed that the gateway does not announce itself on the network. In other words:
•	No address claim at power-up.
•	No answer to 60928, 126996, and 126998 PGN requests from other nodes.
•	No heartbeat broadcast.
Could you please provide an example that implements these behaviors?
I’m working with an app that sends standard and custom PGNs to the boat's instrumentation, and the gateway should show up on the source MDF selection lists for several data items.
If someone else has already made this type of solution public, could you point me in the right direction?
Thank you!

FarFarSouth, 2023-10-20:
Hi again.
I found the ArduinoGateway example!
Being a bit lazy today :-)
Thanks anyway.

closed 2023-10-20 by FarFarSouth

ttlappalainen, 2023-10-21:
ActisenseListenerSender.ino is full passthru and works best with my NMEA Simulator (see https://kave.fi/Apps/), since it can do adress claiming etc. Simulator is also far more better analyzing too than Actisense Reader. Unfortunately messages will be shown parsed only in paid version.