ttlappalainen/NMEA2000 #359: enum tN2kTempSource

greenhouse69, 2023-10-23:
I have read the documentation, but I have some doubts. To define 'tN2kTempSource' as 'SeaTemperature,' what would be the correct way?

void SendN2kTemp(double temp) {
SetN2kEParameters(N2kMsg, 255, N2kts_SeaTemperature, (temp), N2khs_Undef, N2kDoubleNA, N2kDoubleNA); 
NMEA2000.SendMsg(N2kMsg);
 }

or

void SendN2kTemp(double temp) {
int VTempSource = 0
SetN2kEParameters(N2kMsg, 255, (VTempSource), (temp), N2khs_Undef, N2kDoubleNA, N2kDoubleNA); 
NMEA2000.SendMsg(N2kMsg);
 }
Or is there any other way?

ttlappalainen, 2023-10-24:
What do you mean - there is no function SetN2kEParameters?

E.g., for SetN2kTemperatureExt proper way is:
SetN2kTemperatureExt(N2kMsg,N2kUInt8NA,0,N2kts_SeaTemperature,temp);

Note that temp must be in SI units (K).

svdrummer, 2023-10-24:
I wonder why they chose Kelvin scale. instead of Celsius. I could understand out in space. Most sensors are in degrees C. I know it is just floating point math.

ttlappalainen, 2023-10-24:
In general it is best to keep inside code all values in SI and then use conversions for showing data in format user likes. Note that in US they want to keep all temperatures in F - also some US devices provides temperature in F.

Library offers several ready conversion functions like CToKelvin and also FToKelvin for US sensors.

greenhouse69, 2023-10-24:
I don't think my question was understood. The idea is to pass 'tN2kTempSource' as 'SeaTemperature' through a variable, like so: (N2kMsg, 255, 0, (temp), N2khs_Undef, N2kDoubleNA, N2kDoubleNA);"

This code snippet suggests passing the 'tN2kTempSource' enumeration value 'N2kts_SeaTemperature' as an argument to a function or variable.
int N2kts_SeaTemperature = 0
(N2kMsg, 255, (N2kts_SeaTemperature), (temp), N2khs_Undef, N2kDoubleNA, N2kDoubleNA)

greenhouse69, 2023-10-24:
enum  	tN2kTempSource {
  N2kts_SeaTemperature =0 , N2kts_OutsideTemperature =1 , N2kts_InsideTemperature =2 , N2kts_EngineRoomTemperature =3 ,
  N2kts_MainCabinTemperature =4 , N2kts_LiveWellTemperature =5 , N2kts_BaitWellTemperature =6 , N2kts_RefridgerationTemperature =7 ,
  N2kts_HeatingSystemTemperature =8 , N2kts_DewPointTemperature =9 , N2kts_ApparentWindChillTemperature =10 , N2kts_TheoreticalWindChillTemperature =11 ,
  N2kts_HeatIndexTemperature =12 , N2kts_FreezerTemperature =13 , N2kts_ExhaustGasTemperature =14 , N2kts_ShaftSealTemperature =15
}

ttlappalainen, 2023-10-24:
Which code snipped? N2kts_SeaTemperature is enumaration value of type tN2kTempSource and proper way is to provide it for function parameter or variable of type tN2kTempSource. In your example you have redifined N2kts_SeaTemperature  to int and I wonder doesn't compiler warn that. You should call function:
SetN2kEnvironmentalParameters(N2kMsg, N2kUInt8NA, N2kts_SeaTemperature, temp);
without redefination int N2kts_SeaTemperature = 0. If you want to pass it by variable you can provide source e.g., as your sender function parameter:

void SendN2kTemp(tN2kTempSource TempSource, double temp)  {
  SetN2kEnvironmentalParameters(N2kMsg, N2kUInt8NA, TempSource, temp);
  ...
}

Compiler will accept (may show warning) also:
SetN2kEnvironmentalParameters(N2kMsg, N2kUInt8NA, 0, temp);
but this is not so readable - code reader must know that 0 means sea temperature. Compiler will in this case do automatic type casting from int to tN2kTempSource.

greenhouse69, 2023-10-24:
Completely agree with: 
void SendN2kTemp(tN2kTempSource TempSource, double temperature) {
SetN2kEnvironmentalParameters(N2kMsg, N2kUInt8NA, TempSource, temperature);
...
} 
but my question is that I don't know what values to assign to the variable TempSource. 

To clarify, TempSource should take the values: 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, or 15.

or does it take the values: N2kts_SeaTemperature, N2kts_OutsideTemperature, N2kts_InsideTemperature, N2kts_EngineRoomTemperature, N2kts_MainCabinTemperature, N2kts_LiveWellTemperature, N2kts_BaitWellTemperature, N2kts_Refridge Temperatura de ración, N2kts_HeatingSystemTemperature ,N2kts_DewPointTemperature, N2kts_ApparentWindChillTemperature, N2kts_TheoreticalWindChillTemperature,N2kts_HeatIndexTemperature, N2kts_FreezerTemperature, N2kts_ExhaustGasTemperature, N2kts_ShaftSealTemperature.
If I use these values:N2kts_SeaTemperature, N2kts_OutsideTemperature, N2kts_InsideTemperature...,  it works, but I don't know if the library expects numeric values.This is my doubt.

ttlappalainen, 2023-10-25:
As I wrote above use enum value definitions N2kts_SeaTemperature, N2kts_OutsideTemperature...

DO NOT USE 0,1,2,... for TempSource!

It is not library specific way. If function parameter or variable is enum type, it expects enum value definition - NOT INTEGER. Please study e.g., https://en.cppreference.com/w/cpp/language/enum

greenhouse69, 2023-10-26:
ok thank you. I just wanted to understand why it was equated to the numbers 0, 1, 2, 3, 4,...