ttlappalainen/NMEA2000 #360: j1939cat vs nmea2000 iso-tp

skarlsson, 2023-10-25:
This is probably a stupid question - but is iso-tp in j1939 & nmea2000 identical? I would like to make an OTA using can and found the "j1939cat" tool in linux/can-utils package - this looks like it's possible to "stream" a large binary file (maybe in chunks) and feed a normal arduino ota (that usually expects a stream from serial or wifi). If this "should" be possible - how much of this is already handled by this library?  and any pointers where to start would be appreciated.

ttlappalainen, 2023-10-25:
NMEA2000 ISO TP has been inherited from J1939. Unfortunately currently library handles max. 233 packages instad of ISO TP >1800 bytes. Handling is anyway automatic. If library receives ISO TP message, it will be available on call back as any message (tN2kMsg). If it is possible to limit package size, it could work.

skarlsson, 2023-10-26:
Still looking though code - but it seems like j1939 iso tp (maybe for docan) uses different pgn 0x18DA vs 060160 for nmea2000 - I have some old code for 1939 that I will adapt for test purposes - but I did not find any example on how to send a iso tp request (from nmea2000 code). is sending a request implemented or just the receive part?

ttlappalainen, 2023-10-27:
60160 is J1939 definition for ISO TP data transfer. I do not know where did you got  0x18DA.

With my NMEA Simulator you can send messages with ISO TP, but you need commercial box for it. See kave.fi/Apps

skarlsson, 2023-10-27:
Interesting, you are most likely right, I took those id's from code working on jk1939 on actual trucks that supposedly used iso-tp to do UDS (diagnostics) 

ref https://support.vector.com/kb?id=kb_article_view&sysparm_article=KB0011545&sys_kb_id=77d3748887ed01d0b9f233770cbb35ea&spa=1. 

Anyhow, I have what I need - thanks for the support (closing the issue)

ttlappalainen, 2023-10-30:
N2k can carry in optimum about 2200 frame/s. Library seem to limit data transfer to 5 data frame/time. 200 bytes needs 29 data frames and due to limited to 5 data frame it also requires 6 CTS. So totally RTS+29 data+6 CTS+ACK=37 frames. 100 kB requires 100000/200 = 500 messages , * 37 frames = totally 18500 frames. So at optimal time would be 18500 frames / 2200 frame/s = 8.4 s . Check my calculations. Anyway difference is order of 20.

You could try to change on NMEA2000.cpp line 115 #define TP_MAX_FRAMES 5 to e.g., 20 to see if that helps anything.

Problem has to be anyway on handshaking. System has to send CTS 500*6 + ACK 500 times = 3500 times. have you checked you do not have any delay on you loop call to tNMEA2000::ParseMessages()? That takes care of messages construction. I have seen people to think that there is no hurry to call that and having there 10-100 ms delay (or some task timing). With 10 ms you would cause 35000 ms extra delay.

How about sender side? How fast it responds to CTS?

You should have some bus analyzer tool (like my NMEA Simulator) to record data frame timing, so you could see what happens on the bus. It also may be that sending data 5 frames/time could have time to time error and cause request them again. So sending 20 at time can make it even worse. This you could also see with my Simulator.

skarlsson, 2023-10-30:
You were right again! there was a sleep 10ms in my linux code when the "can rx stream" was empty - when changed to 1ms I got to 6kB/s. also changing 5 -> 20 in nmea2000.cpp got me 7.7kB/s. But 6 kB/s is good - and it will be slower in real life because I'm not yet writing to flash. 

many thanks!

ttlappalainen, 2023-10-30:
Nice to be right:)

Had to check theory and calculations. Full frame size can vary somwhere between 128 and 144 bits (did not dig this deeper) and then there may be 3 bits between frames on the bus. So good approximate for optimal speed would be 1800 frames/second. Now you need for single 200 B messages RTS+29 data+2 CTS+ACK=33 frames. So max speed would be 200 B * 1800 frames/s / 33 frames = 10.9 kB/s. With 7.7 kB/s you are near theoretical speed. In Linux you can use <1 ms delay - I used in some tests 100 ns delay. If I remember right in ESP32 <1 ms delay does not let other tasks to run and may cause watchdog.