ttlappalainen/NMEA2000 #361: List of messages not passed to message handlers, possibility to change/override.

SW-Dev-Jay, 2023-10-30:
Hi Timo,
First of all , thanks for all excellent work put into this library, it has proven very useful to me in creating a multitude of different NMEA2000 applications!

Aside the above, I'm now creating a Teensy 4.0 network bridge/filter application for integrating a sensor/MFD combination that is not working properly due to manufacturer handling of data fields set to N/A. Basically I identify a specific PGN message on CAN1 (the sensor side), parse a data field, modify it and send it along on CAN2 (the main N2k network side). 
I have been successful in implementing this using two NMEA2000 instances in ListenAndSend mode, one for each CAN port on the Teensy 4.0 and implementing needed modification of data field in a message handler. All other PGNs will be retransmitted without modification, bidirectionally CAN1 <-> CAN2, also using message handlers.
The ListenAndSend mode means the bridge/filter is invisible to both networks but the way messages are passed to message handlers I seem to miss some network handling messages for the sensor to show up in MFD Device Listings, get the sensor to be configured by the MFD etc. Since I´d want the main network side to be fully aware of the sensor and be capable to interact with it I would also want to have those messages pass though the bridge/filter. How is this implemented in best way using your library?

/Jay

ttlappalainen, 2023-10-30:
Mode naming is not the best. N2km_ListenAndSend mode just automatically forwars messages to ForwardStream. If you have not defined ForwardStream or you have set forward false it is equal as N2km_SendOnly mode. In worst case if you have forwarding on, it may even slow down system. 

MessageHandler will be called in all modes.

I would expect it should work with:
```
void MsgHandlerCAN1(const tN2kMsg &N2kMsg) {
  if ( N2kMsg.PGN==<Specifig PGN> ) {
    tN2kMsg N2kMsgModified(N2kMsg.Source); // Priority, PGn and data length should come from message setter function.
   // Create modified message
   NMEA2000CAN2.SendMsg(N2kMsgModified,-1);
  } else {
    NMEA2000CAN2.SendMsg(N2kMsg,-1);
  }
}

void MsgHandlerCAN2(const tN2kMsg &N2kMsg) {
  NMEA2000CAN1.SendMsg(N2kMsg,-1);
}
```
Here it is important to use -1 as device index on sending so that message is really pass trough. If you use default, it will force sender source to your device source.

SW-Dev-Jay, 2023-10-30:
Thanks Timo,

**Real passthrough of unmodified messages:**
The -1 argument solved the issue. I was unable to comprehend that from reading the NMEA2000.cpp source code, thanks!

**Mode ListenAndSend;** 
I will probably want to use forwarding to USB in Actisense format as to have the unit double up as both filter/bridge and NW monitor using Actisense Reader PC SW. The Teensy 4.0 has good horsepower and I assume it will keep up with these workloads.

For clarity and for others to find correct topic, should I change title of this issue to something that is more accurate? (this had nothing to do with what messages are passed to message handlers)

Again, thanks!
/Jay

ttlappalainen, 2023-10-30:
I'll try to remember to add -1 note to SendMsg comment, which then goes to document. It is a bit hard to read from code.
`
if (DeviceIndex>=0) { N2kMsg.ForceSource(Devices[DeviceIndex].N2kSource); } else { DeviceIndex=0; }`

So for device definition >=0 it forces source to be changed device source. Otherwice source will be untouched, but device would be forced to be 0 (pass through device).

Teensy 4.0 is powerfull. You can extend USB buffers to be used by changing NUM_USB_BUFFERS definition on cores\teensy4\usb_desc.h. Or as I did changed definition to
  #if !defined(NUM_USB_BUFFERS)
  #define NUM_USB_BUFFERS	12
  #endif
and then added platformio.ini build flags -DNUM_USB_BUFFERS=30

For analyzing bus my NMEA Simulator is far more better than Actisense Reader. It shows messags in list or sequential, what AR does not do at all. It has filters and highlighting. Unfortunately not free for showing messages parsed.

SW-Dev-Jay, 2023-10-30:
Thanks for tip on Teensy 4.x USB buffers, I will probably include that and keep track of any changes that come out from PJRC.
Closing this issue as it is was quickly resolved. Cheers!

SW-Dev-Jay, 2023-10-30:
Resolved

closed 2023-10-30 by SW-Dev-Jay