ttlappalainen/NMEA2000 #364: Logging to Log file

RemcoHalman, 2023-11-23:
Hi Timo, 

Could you give a look at the following script i have? I want to record specific data in a Handler (PGN127501)

What I'm trying to do is run over the status and setting it to a human readable format and print them out. Added is also the Write SD function so you see what that does. 

I'm writing it to a log file. And want to change this later on to a CSV file. This would later on be implemented in a custom software package for service. So have been looking at [this](https://github.com/AK-Homberger/NMEA2000-VoyageDataRecorder) as a reference but this writes it to Actisense, Yachtdevices etc. Not a a regular Log/CSV file.

I you have the time much appreciated!

//***************************** Binary Reader *********************************
void HandleBinaryRead(const tN2kMsg &N2kMsg) {
  unsigned char BankInstance;
  tN2kBinaryStatus BankStatus;
  
  if (ParseN2kBinaryStatus(N2kMsg, BankInstance, BankStatus)) {
    String status[28];
    Serial.printf("PGN127501: Binary status for bank %i : ", int(BankInstance));
    for (uint8_t i = 1; i <= 28; i++) {
      if (i > 1) Serial.print(",");
      int switchIndex = N2kGetStatusOnBinaryStatus(BankStatus, i);
      String StateIndex;
      switch (switchIndex)
      {
        case 0: StateIndex = "Off"; break; // No, Off, Disabled
        case 1: StateIndex = "On"; break; // Yes, On, Enabled
        case 2: StateIndex = "Error"; break; // Error
        case 3: StateIndex = "NA"; break; // Unavailable
      }
      Serial.print(StateIndex);
      status[i] = StateIndex;
    }
    // for (int i = 0; i < sizeof(status); i++) Serial.print(status[i]);
    // String to SD;
    char text[150];
    writeSD(text);
    Serial.println();
  }
}`

`
//************************ SD card message handler ****************************
void writeSD(const char *message) {
  const char *path = "/data/data.log";
  File file = SD.open(path);
  if (!file) {
    Serial.println("Creating file for writing");
    File file = SD.open(path, FILE_WRITE);
    if (file.print(message)) {
      // Serial.println("Message writen");
    }
  } else {
    File file = SD.open(path, FILE_APPEND);
    if (file.print(message)) {
      // Serial.println("Message appended");
    }
  }

  digitalWrite(LedOnTime, LedState);
  LedState = !LedState;
}`

ttlappalainen, 2023-11-24:
What was your question? How to construct message to be saved?

At least for writing log I'll prefer to ad first time column and save previous state. Then you can log only changes. E.g.,
  tN2kBinaryStatus BankStatus;
  static tN2kBinaryStatus PreviousBankStatus;

  if (ParseN2kBinaryStatus(N2kMsg, BankInstance, BankStatus)) {
    if ( PreviousBankStatus!=BankStatus ) {
      PreviousBankStatus=BankStatus;
      ...
      writeSD(text);
    }
  }

RemcoHalman, 2023-11-24:
Hi Timo thanks for the response. If you could give a example of hoe to construct the message?

I would prefer that as well but the "problem" im facing is multiple bank instances. for example the values below are from 2 different tanks levels. How would you write it to a buffer/construct to log?

this would save a ton of space on a SD card because only a small script is running 100K lines a hour with only 7 PGN's. That's not including GPS.

```
PGN127505: instance: 12, type: Fuel, Level: 64.6% 
PGN127505: instance: 10, type: Water, Level: 80.4% 
```

```
//*****************************************************************************
//**************************** Fluid Handler **********************************
//*****************************************************************************
void HandleFluid(const tN2kMsg &N2kMsg) {
  unsigned char Instance;
  tN2kFluidType FluidType;
  String FluidData;
  int InstanceUsed;
  double Level;
  double Capacity;
  double Liters;

  if (ParseN2kPGN127505(N2kMsg, Instance, FluidType, Level, Capacity)) {
    InstanceUsed = int(Instance);
    switch (FluidType) {
      case N2kft_Fuel: FluidData = "Fuel"; break;
      case N2kft_Water: FluidData = "Water"; break;
      case N2kft_GrayWater: FluidData = "Gray water"; break;
      case N2kft_LiveWell: FluidData = "Live well"; break;
      case N2kft_Oil: FluidData = "Oil"; break;
      case N2kft_BlackWater: FluidData = "Black water"; break;
    }
    if (Capacity != 0) {
      Liters = Capacity * Level / 100;
    }
    if (Level > 101 || Level < -1) {
      Level = 0;
    }
    // String to SD;
    Serial.printf("PGN127505: instance: %i, type: %s, Level: %4.1f%%, Capacity: %4.1fL \n", InstanceUsed, FluidData, Level, Capacity);
    char text[100];
    sprintf(text, "PGN127505: instance: %i, type: %s, Level: %4.1f%%, Capacity: %4.1fL\n", InstanceUsed, FluidData, Level, Capacity);
    writeSD(text);
  }
}

```

ttlappalainen, 2023-11-24:
It is not so simple. It depends of your log purpose. 
- If you want to log all sentences, it may be best to log them as hex with timestamp. Then you need separate program to read log. I log all messages with OpenSkipper to computer, but half day log is about 200 MB. I can read log later with my NMEA Simulator.
- Or you can log in csv format. In this case you do not need to repeat text like instance, Level for every row. But on the other hand, if you log different PGNs, you need own column for each data and each PGN should write empty data for columns they do not have data. csv is good, ig you want to read it with e.g., Excel. Example
Time,PGN,instance,fluid type, level,temperature source,temperature
2023-11-24T12:30:30.500,127505,12,0,64.6,,
2023-11-24T12:30:31.500,127505,10,1,80.4,,
2023-11-24T12:30:32.500,130316,,,,4,294.15
- One way is to collect data and log it periodically to csv so that every line has all data. By using longer period, you can reduce log size.
- If you want to have it human readable each line you have do it as in your example.
- If it is not important to log every message, you need to save values for each PGNs and log on, when they change. For real values like tank level you may need to use some hysteresis.
- Or you can do combination and log to csv all values, if any of them changes.

RemcoHalman, 2023-11-26:
Hi Timo, 

Sorry for the late response. 

It's not required for all data to be saved. I want only selected. Just like the Fluid i have more handlers in place. I'm no expert in C++ so still learning. I think the file size is still ok if it's less then 20GB or so.

Could you give a small example how to write the binary read to a single buffer and print that out? The most other PGN's are perfectly doable with the predefined parsers. But the binary swithing is in a for loop. And im not able to set them to a array... I know this is my own issue but would love some help. 

The idea is like you said to create a different progam to read it. But also setting them in charts for example the fluid. The basic timestamp wil be the millis run. The system time will be a filter to select different dates. Todo there is set a config file and update the millis say every 10 seconds. That way if the device reboot or the boat restarts the milis is somewhat saved. But thats a different story.

RemcoHalman, 2023-11-27:
For now i solved it doing the following piece of code If you have any tips or tricks to add/change please let me know.

```
//*****************************************************************************
//***************************** Binary Reader *********************************
//*****************************************************************************
char *array_to_str(char * str, int *array, unsigned int n) {
  int r;
  if (n == 0) return 0;
  if (n == 1) r = sprintf(str, "%d", array[1]);
  else        r = sprintf(str, "%d, ", array[1]);
  array_to_str(str + r, array + 1, n - 1); 
  return str;
}

void HandleBinaryRead(const tN2kMsg &N2kMsg) {
  unsigned char BankInstance;
  tN2kBinaryStatus BankStatus;
  int array_size = 28;
  char buffer[250];

  
  if (ParseN2kBinaryStatus(N2kMsg, BankInstance, BankStatus)) {
    #ifdef DEBUG_SWITCHING
    int values[array_size + 1];
    for (uint8_t i = 1; i <= array_size; i++) {
      int switchIndex = N2kGetStatusOnBinaryStatus(BankStatus, i);
      values[i] = switchIndex;
    }
    char text[150];
    sprintf(text, "PGN127501: Binary status for bank %i : %S \n", int(BankInstance), array_to_str(buffer, values, array_size));
    Serial.print(text);
    // String to SD;
    writeSD(text);
  #endif
  }
}
```

ttlappalainen, 2023-11-27:
A bit complex solution. Your routine will do recursive call, which is not good with small MCU:s.  Also it is safer (always) to use snprintf, since it does not overwrite buffer. Simple would be:
#define MAX_BITS 28
void N2kBinaryBankToString(tN2kBinaryStatus BankStatus, char *buf, size_t nBuf) {
  char *tmpBuf=buf;
  int nWritten;
  size_t nTmpBuf=nBuf;

  *tmpBuf=0;
  for ( int i=0; i<MAX_BITS; i++ ) {
    nWritten=snprintf(tmpBuf,nTmpBuf,"%u",(uint8_t)(BankStatus & 0x3));
    if ( nWritten>0 && nWritten<nTmpBuf ) {
      nTmpBuf-=(size_t)nWritten;
      tmpBuf+=nWritten;
      BankStatus>>=3;  // Shift to next binary bit.
    } else {
      break; // Buffer full or failure. We have to stop.
    }
  }  
}

ttlappalainen, 2023-11-27:
You must have char binaryBuffer[MAX_BITS+1]; , since strings are 0 terminated and need one extra byte for that.

closed 2023-11-28 by RemcoHalman