ttlappalainen/NMEA2000 #365: Ambiguous documentation section

trixpan, 2023-11-25:
Reading the following section it is unclear what is this supposed to mean

> Devices with other isolated sensors like pressure, ultrasonic, hall effect current etc. Those must get power from your board and their body must be isolated from their ground.

Their body refers to the device or sensors bodies? Same about ground. What ground?

My understanding is that you are trying to convey  that as long the external sensors are gounded to the board and the board grounded to the bus with no further external grounds, the non isolated solution can be used. Is this correct?

ttlappalainen, 2023-11-26:
Not exactly. Sorry my English, please send me how it should be written to be clear so I can update it.

Body means sensor body and ground means board ground. So if sensor body is isolated from board ground, one can use non isolated solution. In that case board ground and also NMEA2000 bus ground does not connect to the vessel ground through body. If sensor body has connection any sensor signal (normally ground) ans you screw sensor e.g., to engine, then you cause ground loop through engine and sensor to bus. 

Also should be mentioned that in case one has long sensor wires, they may start to work as antennas and cause other disturbance to bus. Here one may think that there is no difference if bus drop cable is longer and sensor cable shorter. The difference is that bus cable has shield.

Isolated solution is always safest solution.

trixpan, 2023-11-26:
@ttlappalainen I will be happy to suggest some content when the meaning is a clearer between both of us. We will get there.

So if I get it right, by "sensor body isolation" you are referring to the need to either isolate the MCU to sensor wires (e.g. using a bus buffer like a P82B96)  to prevent the possibility of any path allowing electricity to reach ground that is not the NET-C?

Is this correct?

ttlappalainen, 2023-11-26:
No. P82B96 is just buffer to allow of using long cables with I2C. It is not isolating device. If you use P82B96 and your sensor body is connected to ground wire, you have grounded NET-C by sensor body.

Isolation can be done by either optoisolator or capacitive isolator as in ISO1050.

trixpan, 2023-11-26:
@ttlappalainen afaik you can use the P82B96 for isolating the i2c bus, this is documented here:

https://www.nxp.com/docs/en/application-note/AN10364.pdf

This in theory, assuming there are a different Vcc and GND - as one would expect when using an optoisolator as well - should grant isolation of the MCU from the sensor body, while still allowing the MCU to be connected to the CAN.

And yes, I do agree using a pure optoisolator or a capacitive isolators are purpose build, easier and safer choice. All I was trying to is to hone in, into what you meant by sensor body isolation.

My understanding so far is that the above mentioned section of the text is basically trying to emphasise that since isolation isn't in place under that suggested design, extreme care must be taken so that not even conductive sections of the body of the sensor touch any other conductive material that may route to ground.

ttlappalainen, 2023-11-26:
OK. But I do not see yor point to make things very complex. If one needs isolation, why not then just use isolated tranceiver? And if device should be powered from bus, then use also isolated DC-DC converter?

Note also that the sensor is not necessarily I2C. It may be 1-wire, AC, etc. If one wants to read simple engine temperature sensor by using MCU ADC, one must use isolated connection to bus. If engine temperature sensor is 2 wire with isolated body, then isolation is not required.

trixpan, 2023-11-26:
@ttlappalainen I am not seeking advise on how to do things, nor suggesting a complex design, all I am trying to do is to understand what you originally meant so I can write something that is clearer to the next reader. :-) 

But yes, I think we are now on the same page. Will submit a PR later today. :-)

trixpan, 2023-12-04:
@ttlappalainen hopefully the PR linked above is correct.

PR #367 (2023-12-08): Update 3_hwSetup.md
Closes #365
commit 3940819b (2023-12-08): Update 3_hwSetup.md

Closes #365