ttlappalainen/NMEA2000 #368: tN2kGNSSDOPmode ActualMode,

Workcar1, 2023-12-06:
good time of day. I am sorry for my English. The task is to transfer the GNGSA parameters to H2k. ParseGSA is running, data has been received. How to transfer 2D/3D data to tN2kGNSSDOPmode ActualMode,

void HandleGSA(const tNMEA0183Msg &NMEA0183Msg) {  
     if (pBD==0) return;

   if (NMEA0183ParseGSA_nc(NMEA0183Msg,pBD->DesMode,pBD->ActMode,                                         // ActMode--------- режим 2d/3d
                         pBD->ID1, pBD->ID2, pBD->ID3, pBD->ID4, pBD->ID5, pBD->ID6, pBD->ID7, pBD->ID8,
                         pBD->ID9, pBD->ID10, pBD->ID11, pBD->ID12,
                       pBD->PDOP, pBD->HDOP1, pBD->VDOP1)) {
                  
     if (pNMEA2000!=0) {
    tN2kMsg N2kMsg;
   SetN2kGNSSDOPData(N2kMsg, 25, N2kGNSSdm_3D,pBD->ActMode, pBD->HDOP1, pBD->VDOP1,0);         // N2kGNSSdm_Auto
   pNMEA2000->SendMsg(N2kMsg);
     }     
  }
}
ERROR
initializing argument 4 of 'void SetN2kGNSSDOPData(tN2kMsg&, unsigned char, tN2kGNSSDOPmode, tN2kGNSSDOPmode, double, double, double)'
 inline void SetN2kGNSSDOPData(tN2kMsg& N2kMsg, unsigned char SID, tN2kGNSSDOPmode DesiredMode, tN2kGNSSDOPmode ActualMode,
                                                                                                ~~~~~~~~~~~~~~~~^~~~~~~~~~
exit status 1
invalid conversion from 'int' to 'tN2kGNSSDOPmode' [-fpermissive]

Best regards, Dmitry

Workcar1, 2023-12-10:
All options. There is no 3D output.

// ******              INPUT           *************              OUT   NMEA READER --- effect
  // N2kGNSSdm_1D  == 0     N2kGNSSdm_1D  == 0    result    Set mode == 0 (1D Fix)    Op mode == 0 (1D Fix)
  // N2kGNSSdm_1D  == 0     N2kGNSSdm_2D  == 1    result    Set mode == 4 (Reserved)  Op mode == 0 (1D Fix)
  // N2kGNSSdm_1D  == 0     N2kGNSSdm_3D  == 2    result    Set mode == 0 (1D Fix)    Op mode == 1 (2D Fix)
  // N2kGNSSdm_1D  == 0     N2kGNSSdm_Auto== 3    result    Set mode == 4 (Reserved)  Op mode == 1 (2D Fix)
  
  // N2kGNSSdm_2D  == 1     N2kGNSSdm_1D  == 0    result    Set mode == 0 (1D Fix)    Op mode == 4 (reserved)
  // N2kGNSSdm_2D  == 1     N2kGNSSdm_2D  == 1    result    Set mode == 4 (Reserved)  Op mode == 4 (reserved)
  // N2kGNSSdm_2D  == 1     N2kGNSSdm_3D  == 2    result    Set mode == 0 (1D Fix)    Op mode == 5 (reserved)
  // N2kGNSSdm_2D  == 1     N2kGNSSdm_Auto== 3    result    Set mode == 4 (Reserved)  Op mode == 5 (reserved)
  
  // N2kGNSSdm_3D  == 2     N2kGNSSdm_1D  == 0    result    Set mode == 0 (1D Fix)     Op mode == 0 (1D Fix)
  // N2kGNSSdm_3D  == 2     N2kGNSSdm_2D  == 1    result    Set mode == 4 (Reserved)   Op mode == 0 (1D Fix)
  // N2kGNSSdm_3D  == 2     N2kGNSSdm_3D  == 2    result    Set mode == 4 (Reserved)   Op mode == 0 (1D Fix)
  // N2kGNSSdm_3D  == 2     N2kGNSSdm_Auto== 3    result    Set mode == 4 (Reserved)   Op mode == 0 (1D Fix)

  // N2kGNSSdm_Auto== 3     N2kGNSSdm_1D  == 0    result    Set mode == 0 (1D Fix)     Op mode == 4 (Reserved)
  // N2kGNSSdm_Auto== 3     N2kGNSSdm_2D  == 1    result    Set mode == 4 (Reserved)   Op mode == 4 (Reserved)
  // N2kGNSSdm_Auto== 3     N2kGNSSdm_3D  == 2    result    Set mode == 0 (1D Fix)     Op mode == 5 (reserved)
  // N2kGNSSdm_Auto== 3     N2kGNSSdm_Auto== 3    result    Set mode == 4 (Reserved)   Op mode == 5 (reserved)
   
   SetN2kGNSSDOPData(N2kMsg, 25, N2kGNSSdm_Auto, N2kGNSSdm_Auto, pBD->HDOP1, pBD->VDOP1,0);   
   
     Serial.print("N2kGNSSdm_Auto  -----------------  ");Serial.println(N2kGNSSdm_Auto);  
     Serial.print("N2kGNSSdm_Auto  -----------------  ");Serial.println(N2kGNSSdm_Auto);

ttlappalainen, 2023-12-10:
On N2kMessages.cpp line 1227 is wrong. It should be:
    N2kMsg.AddByte( 0xc0 | ( (ActualMode & 0x07) << 3 ) | (DesiredMode & 0x07) );

Also lines 1244 and 1245 should be
    DesiredMode = (tN2kGNSSDOPmode)( modes & 0x07 );
    ActualMode = (tN2kGNSSDOPmode)( (modes>>3) & 0x07 );

Workcar1, 2023-12-10:
Timo Lappalainen YOU are a great person. Thanks a lot.