ttlappalainen/NMEA2000 #370: N2kMaretron.h compile error

juspa1978, 2023-12-22:
In file included from C:\Users\juspa\iCloudDrive\arduino\nmea2000\M5_NMEA_Display_CAN_Bus\NMEA2000_esp32.h:7,
                 from C:\Users\juspa\iCloudDrive\arduino\nmea2000\M5_NMEA_Display_CAN_Bus\NMEA2000_esp32.cpp:33:
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:79:94: error: 'tN2kTempSource' has not been declared
 void SetN2kMaretronPGN130823(tN2kMsg &N2kMsg, unsigned char SID, unsigned char TempInstance, tN2kTempSource TempSource,
                                                                                              ^~~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:89:98: error: 'tN2kTempSource' has not been declared
 inline void SetN2kMaretronTempHR(tN2kMsg &N2kMsg, unsigned char SID, unsigned char TempInstance, tN2kTempSource TempSource,
                                                                                                  ^~~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:127:104: error: 'tN2kTempSource' has not been declared
 bool ParseN2kMaretronPGN130823(const tN2kMsg &N2kMsg, unsigned char &SID, unsigned char &TempInstance, tN2kTempSource &TempSource,
                                                                                                        ^~~~~~~~~~~~~~
In file included from C:\Users\juspa\iCloudDrive\arduino\nmea2000\M5_NMEA_Display_CAN_Bus\NMEA2000_esp32.h:7,
                 from C:\Users\juspa\iCloudDrive\arduino\nmea2000\M5_NMEA_Display_CAN_Bus\NMEA2000_esp32.cpp:33:
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:138:108: error: 'tN2kTempSource' has not been declared
 inline bool ParseN2kMaretronTempHR(const tN2kMsg &N2kMsg, unsigned char &SID, unsigned char &TempInstance, tN2kTempSource &TempSource,
                                                                                                            ^~~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:171:97: error: 'tN2kFluidType' has not been declared
 void SetN2kMaretronPGN65286(tN2kMsg &N2kMsg, unsigned char SID, unsigned char FlowRateInstance, tN2kFluidType FluidType,
                                                                                                 ^~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:181:103: error: 'tN2kFluidType' has not been declared
 inline void SetN2kMaretronFluidFR(tN2kMsg &N2kMsg, unsigned char SID, unsigned char FlowRateInstance, tN2kFluidType FluidType,
                                                                                                       ^~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:217:107: error: 'tN2kFluidType' has not been declared
 bool ParseN2kMaretronPGN65286(const tN2kMsg &N2kMsg, unsigned char &SID, unsigned char &FlowRateInstance, tN2kFluidType &FluidType,
                                                                                                           ^~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:227:113: error: 'tN2kFluidType' has not been declared
 inline bool ParseN2kMaretronFluidFR(const tN2kMsg &N2kMsg, unsigned char &SID, unsigned char &FlowRateInstance, tN2kFluidType &FluidType,
                                                                                                                 ^~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:259:95: error: 'tN2kFluidType' has not been declared
 void SetN2kMaretronPGN65287(tN2kMsg &N2kMsg, unsigned char SID, unsigned char VolumeInstance, tN2kFluidType FluidType,
                                                                                               ^~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:269:104: error: 'tN2kFluidType' has not been declared
 inline void SetN2kMaretronTripVolume(tN2kMsg &N2kMsg, unsigned char SID, unsigned char VolumeInstance, tN2kFluidType FluidType,
                                                                                                        ^~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:304:105: error: 'tN2kFluidType' has not been declared
 bool ParseN2kMaretronPGN65287(const tN2kMsg &N2kMsg, unsigned char &SID, unsigned char &VolumeInstance, tN2kFluidType &FluidType,
                                                                                                         ^~~~~~~~~~~~~
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h:314:114: error: 'tN2kFluidType' has not been declared
 inline bool ParseN2kMaretronTripVolume(const tN2kMsg &N2kMsg, unsigned char &SID, unsigned char &VolumeInstance, tN2kFluidType &FluidType,
                                                                                                                  ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

ttlappalainen, 2023-12-22:
N2kMaretron.h should have #include "N2kTypes.h" after #include "N2kMsg.h". You can temporaly fix problem by adding
#include <N2kTypes.h>
before including N2kMaretron.h.
I have added fix to next update.

ttlappalainen, 2023-12-22:
Actually I wonder why it complains:
In file included from C:\Users\juspa\iCloudDrive\arduino\nmea2000\M5_NMEA_Display_CAN_Bus\NMEA2000_esp32.h:7,
from C:\Users\juspa\iCloudDrive\arduino\nmea2000\M5_NMEA_Display_CAN_Bus\NMEA2000_esp32.cpp:33:
c:\Users\juspa\OneDrive\Tiedostot\Arduino\libraries\NMEA2000\src/N2kMaretron.h

NMEA2000_esp32.h has just comments on line 7. Have you done some modifications to that file?

closed 2024-01-21 by ttlappalainen