ttlappalainen/NMEA2000 #371: SID _Instance

greenhouse69, 2023-12-26:
In the case of sending two messages with the same PGN 130316, for example, OutsideTemperature from two different sensors, should I only send each message with a different tempInstance? Is this correct? And in the case of sending multiple messages, for example, with PGN 130313, 130314, and 130316 from the same sensor (bme280), should I send them with the same SID? Is this correct?

ttlappalainen, 2023-12-27:
In most cases you can use just 0xff for SID. I made mistake that I did not put that parameter to last with default value 0xff on parameter list.

SID is defined to make difference between sensor sampling or calculation. Same SID in different PGNs defines that values on PGN:s has been sampled or calculated at same time. SID has some sense if you e.g., have device, which sends wind and speed data or even better also vessel heading. Then in calculations you can use PGNs data provided with same SID and get more accurate results.

With temperature data I do not see any use for it. Temperature changes are normally anyway slow and 1.5-2 s difference on sampling does not have meaningfull effect. I would also think that there is no device using SID for temperature, pressure and humidity data.

If you like to use SID, you have to increase value and run it on range 0-252. 255 (0xff) means data not available and 253-254 are reserved. So use it e.g., with code `SensorSID++; if ( SensorSID>252) SensorSID=0;` If you e.g., sample your multisensor every 2 s, but PGNs will be send every 1 s, there would be 2 sends having same SID indicating that they have been sampled at same time. If you instead sample every 0.5 s, PGNs shows every second SID.

And if we are strict and one use multitasking system, one should also lock data during sampling and sending so that they would not change during task. If you e.g., have task sending sensor values from some kind of globals, both task should use some data locking system:
Sampling task:
```
SampleSensors(Sensor1,Sensor2);
LockSend();
Sensor1Out=Sensor1;
Sensor2Out=Sensor2;
SensorsSID++; if ( SensorsSID>252 ) SensorsSID=0;
ReleaseSend();
```

Sending task;
```
LockSend();
double S1=Sensor1Out;
double S2=Sensor2Out;
SID=SensorsSID;
ReleaseSend();
SendSensorsData(SID,S1,S2);
```

greenhouse69, 2023-12-27:
Thank you, Timo, for your explanations. They are truly helpful for us.