ttlappalainen/NMEA2000 #374: HandleNMEA2000Msg not getting called

superandrew213, 2023-12-30:
I'm using this board https://www.skpang.co.uk/collections/can-bus-boards/products/esp32-can-bus-board.

I just want to log the messages but `HandleNMEA2000Msg` is not getting called.

```cpp
#include <Arduino.h>
#include <N2kMessages.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <NMEA2000_esp32.h>

tNMEA2000_esp32 NMEA2000;

void logByteArray(const byte* byteArray) {
  size_t size = sizeof(byteArray) / sizeof(byteArray[0]);
  for (size_t i = 0; i < size; ++i) {
    Serial.print(byteArray[i], HEX);
    Serial.print(" ");
  }
  Serial.println();
}

// NMEA 2000 message handler
void HandleNMEA2000Msg(const tN2kMsg& N2kMsg) {
  Serial.print("PGN: ");
  Serial.println(N2kMsg.PGN);
  Serial.print("Priority: ");
  Serial.println(N2kMsg.Priority);
  Serial.print("Data: ");
  logByteArray(N2kMsg.Data);
}

void setup() {
  Serial.begin(115200);
  // Do not forward bus messages at all
  NMEA2000.EnableForward(true);
  NMEA2000.SetMsgHandler(HandleNMEA2000Msg);
  NMEA2000.Open();
}

void loop() { NMEA2000.ParseMessages(); }
```

My NMEA2000 network is setup correctly and I can log messages with CanKing.

There are no errors. Am I missing something?

ttlappalainen, 2023-12-30:
You board has CAN_TD and CAN_RD on other pins than tNMEA2000_esp32 constructor defaults. Set them properly before including  NMEA2000_esp32.h as described on documents. E.g.,
```
#define ESP32_CAN_TX_PIN GPIO_NUM_34
#define ESP32_CAN_RX_PIN GPIO_NUM_35
```

Also NMEA2000.EnableForward(true); does not do anything, since you have not set forward stream. You would get nearly same output without handler by just setting:
```
void setup() {
  Serial.begin(115200);
  NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text
  NMEA2000.SetForwardStream(&Serial);
  NMEA2000.EnableForward(true);
  NMEA2000.Open();
}
```
See example ActisenseListener.ino.

superandrew213, 2023-12-30:
Thanks @ttlappalainen. Made those changes. Still nothing.

Tried `ActisenseListener` with pins set properly but still nothing.

Also for anybody else trying this. Pins should be the other way.
```cpp
#define ESP32_CAN_TX_PIN GPIO_NUM_35
#define ESP32_CAN_RX_PIN GPIO_NUM_34
```

ttlappalainen, 2023-12-30:
Those defines were as example. On your board setting should be:
```
#define ESP32_CAN_TX_PIN GPIO_NUM_25
#define ESP32_CAN_RX_PIN GPIO_NUM_26
```

superandrew213, 2023-12-30:
Yes, those are the correct pins. Thanks @ttlappalainen! It's working now.

closed 2023-12-30 by superandrew213