ttlappalainen/NMEA2000 #378: Convert generated output into JSON format

affadli, 2024-01-19:
Hi Timo,
I really appreciated for this awesome library.
I'm using this library for my embedded linux project (socketCan).
All is working well but I'm having issue to convert the data into JSON format since I need to send the data over MQTT.

Do you have any clue for this?

Thanks !

ttlappalainen, 2024-01-19:
Very simple. Look example DataDisplay and there function 
void FluidLevel(const tN2kMsg &N2kMsg) {
Instead of printing data you save FluidType, Level and Capacity to some variable and then on other task put them to Json and send to server. It is no good to sending on function FluidLevel, since that is callback from NMEA2000.ParseMessages() and that should be handled as fast as possible.

With multitasking system do not forget to use mutexes for data locking.

affadli, 2024-01-19:
That's mean I need to construct JSON manually for every PGN. 
It will be nice if when printing the parsed sentences, there is option for JSON format.
So, the data on streamline is JSON already.
This feature very helpful I guess.

closed 2024-01-19 by affadli

ttlappalainen, 2024-01-19:
It is not. NMEA2000 library is independent of any JSON implementation and there is no standard format for NMEA2000 data in Json.