ttlappalainen/NMEA2000 #379: Rx buffer in NodeOnly mode

greenhouse69, 2024-01-22:
A conceptual question, in tNMEA2000::N2km_NodeOnly mode (I understand it is Tx only), shouldn't I implement the Rx buffers:
NMEA2000.SetN2kCANMsgBufSize(6); default 5 messages.
NMEA2000.SetN2kCANReceiveFrameBufSize(250);
And yes, for the TX one:
NMEA2000.SetN2kCANSendFrameBufSize(250);

I have seen some examples, in NodeOnly mode, implementing all 3 buffers:
NMEA2000.SetN2kCANMsgBufSize(6);
NMEA2000.SetN2kCANReceiveFrameBufSize(250);
NMEA2000.SetN2kCANSendFrameBufSize(250);

Which is correct? Thanks.

ttlappalainen, 2024-01-22:
Unfortunately this is very historical mode idea and a bit confusing. The only difference between N2km_NodeOnly and N2km_ListenAndNode is that N2km_NodeOnly does not forward messages. Forward can be disabled with EnableForward(false) which makes N2km_NodeOnly unnecessary.

Only N2km_ListenAndNode and N2km_ListenAndSend are necessary. N2km_ListenAndNode for bus device  and N2km_ListenAndSend for reading bus or fooling.

greenhouse69, 2024-01-23:
I understand that for N2km_NodeOnly mode, RX buffer is not necessary, is that correct? Thank you very much for your help.

ttlappalainen, 2024-01-23:
Yes it is. As it does all required communication with bus as N2km_ListenAndNode, Rx buffer is important. If there is not enough Rx buffer, device tries to work, but works half way. As I said:
"The only difference between N2km_NodeOnly and N2km_ListenAndNode is that N2km_NodeOnly does not forward messages. Forward can be disabled with EnableForward(false) which makes N2km_NodeOnly unnecessary."

greenhouse69, 2024-01-23:
So:

NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 30);

equals

NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode, 30);
NMEA2000.EnableForward(false);

Can be used interchangeably. That N2km_NodeOnly not being necessary doesn't mean that it's better to use N2km_ListenAndNode + EnableForward(false), or are they independent? Thank you for your clarification and your patience.

ttlappalainen, 2024-01-23:
Yes can be used interchangeably before open but not after. SetMode must be called before open and it can not be called after. There actually should be limitation so that it does not do anything after open. As I mentioned, there is some historical reasons.

I personally use
```
NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode, SourceAddress);
NMEA2000.EnableForward(false);
```
With ESP32 I can then command via WEB interface during runtime.
```
NMEA2000.SetForwardStream(UDPStream);
NMEA2000.EnableForward(true);
```
And use the NMEA Simulator for reading bus data.