ttlappalainen/NMEA2000 #380: NMEA2000ToWiFiAsSeaSmart with web server.

greenhouse69, 2024-01-27:
I am testing the NMEA2000ToWiFiAsSeaSmart example. I have modified the code by simply changing the WiFi mode from STA to AP. The issue is that I am trying to implement a web server on the ESP32 to be able to change the IP and TCP port, but I am completely unable to do it. I have tried various libraries (WebServer, ESPAsyncWebServer, and AsyncTCP), but they collide with the existing code. I also tried another very similar example with the NMEA2000ToWiFiAsNMEA0183WithAIS library, and the same issue occurred. Any ideas on how to start the WifiServer server(80) in their example?

ttlappalainen, 2024-01-27:
I have used those libraries WiFi in WIFI_AP mode for configuration. In AP configuration mode I do not start client server. I also use UDP broadcasting so I do not need WiFi server for clients.

greenhouse69, 2024-01-27:
I apologize, I did not understand your response. My intention is to create a web server to easily access the device configuration on my boat. I use WiFi in AP mode because I believe it is the best option on my boat, as I do not have a router installed to connect. That's why I have dismissed the WiFi STA option in your code. I could use the Lowrance plotter, which allows me to create an Access Point, but the truth is that I prefer not to use this option. My intention is to modify your code slightly to start in AP mode, launch the web server (port 80), and TCP on (serverPort). 
WebServer server(80); 
WiFiServer server(serverPort, maxClients). 
My question is, is this possible? I am having many problems trying to implement it.

ttlappalainen, 2024-01-28:
It is possible to have servers to different ports, but there may be problem that some code conflicts and unfortunately I do not have time to test. With your definition
WebServer server(80);
WiFiServer server(serverPort, maxClients);
there should not be problem, if WebServer and WiFiServer classes does not conflict. Sometime those conflicts can be fixed just by changing include order, but can not say in your case.

About WiFi. I have got feeling that ESP32 AP mode is not as good as station mode. I can have several clients using web server on station mode, but in AP mode ESP32 seem to lack for next connection. Also if you would have several of those devices on your boat, you can not connect them simultaneously. I prefer to think investment for good router where all your devices connects and shares same network. I have router with 2.4G and 5G so that computer and cell phone uses 5G and ESPs 2.4G.