ttlappalainen/NMEA2000 #382: How do i announce changes on startup to the network i am connected to ?

KristianJerpetjon, 2024-02-04:
When i keep my nmea2k bus running and change my PGN's is there a nice way to "push" out changes to the network when i update my program ? currently running with an ESP32. seems to me like my "open" should be doing something i am not doing ?  

running a multi device with mode   

NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, NodeAddress);

ttlappalainen, 2024-02-04:
As in example TemperatureMonitor you setup first your device information, product information, transmit/receive PGN list registration  and if you like configuration information. Library takes care of all information to the bus. You just need to take care of your own transmit/receive PGN handling. System PGNs will be handled by the library. Read also "NMEA2000 certification" in the document.

Remember that you must run NMEA2000.ParseMessages() without delays. See in document Running library under Reference for C++ library - NMEA2000 library reference - Using tNMEA2000

KristianJerpetjon, 2024-02-05:
Thanks.. 

I am writing a NKE to N2K bridge. So its mostly Realtime . 

What i am currently trying to figure out is how to emulate a simnet AP so that i can use my BT-1 / WR10  https://www.simrad-yachting.com/simrad/type/autopilots/autopilot-remote-controllers/wr10-remote--bt1-base-station-pack/ wireless controller to control the NKE AP. 

Using the ORCA core2 https://getorca.com/orca-core/ it detects my emulated AP /AC12 however i believe its doing it the wrong way. 

what i have noticed is that the SIMNET commission requests are only issued when i power on the simnet devices. so maybe i have to power cycle the system every time i try. No late net entry possible ? 

Anyhow i think a trip to my B&G/Simnet boat is needed to make progress in decoding this. I guess simnet groups etc also matters.

ttlappalainen, 2024-02-05:
I did not comply understood your problem.
1. If you are making totally simulated device, you need copy all its information from one real device and then use those as data for device, program etc. information. Then you write simulated code as your logic. In that case you do not connect real device to bus at all. May be complex.
2. If you just want to fool data sent like from real device existing on bus, you use library in N2km_ListenAndSend mode. Then you listen real device ISO Address claim and adapt your device using same address. Then if you e.g., like to send +10 command, you send that PGN by using same address as you have received. In this way no-one on bus does not know that PGN was not actually sent from real device and so acts like it was sent from that. That works at least with Raymarine AP, and I can control it from own device. Raymarine actully takes new direction as parameter, so I can command any turn with it.
3. Maybe some other method too.

Have you checked my NMEA Simulator? With that you can send also proprietary messages to the bus and make simple tests. With it you can also do a lot of anayzing with highlighting or filtering.

KristianJerpetjon, 2024-02-05:
Hi i want to simulate an AC12/AC42 or NAC autopilot controller. I think thats what i have to do for it to work with the BR-1/WR10