ttlappalainen/NMEA2000 #384: Support for PEAK Systems devices

phylib, 2024-02-08:
Thank you all for this great NMEA library!

Using the library on ESPs and Arduinos works perfectly fine. Now, there's a point where I would like to use the library on a different device, however. Therefore, I would like to contribute in porting the library for this platform and need advice on how to start.

To be more specific, I would like to use the [PCAN-Router FD](https://www.peak-system.com/PCAN-Router-FD.406.0.html) with the NXP LPC4078 (ARM Cortex M4 mit FPU, 120 MHz). Programming is little different compared to the Arduino framework. Instead of `setup()` and `loop()`, there's only a `main()`. CAN-wise, an adapter for sending and receiving might be neccessary, but shouldn't be a big deal, when knowing where to start.

Can anyone give pointers, on where changes in the NMEA2000 library are neccesary? Or are there already similar ports existing, where I could take a look and adapt it similarly?

Thanks!

ttlappalainen, 2024-02-08:
Library is not fixed to Arduino environment. Used also at least with ESP-IF, RPI, Linux. 

You need to write CAN controllr dependent "driver". There are different CAN driver ports. Good example is NMEA2000_Teensyx or NMEA2000_esp32. See also on document:
- tNMEA2000::CANOpen https://ttlappalainen.github.io/NMEA2000/classt_n_m_e_a2000.html#a7188c4049271a83aa7bd5047412f89ae
- tNEAM2000::CANSendFrame https://ttlappalainen.github.io/NMEA2000/classt_n_m_e_a2000.html#ad91db86761f89ed9f6e88b4f18d4603b
- tNMEA2000::CANGetFrame https://ttlappalainen.github.io/NMEA2000/classt_n_m_e_a2000.html#a551f1437c7bb77848c5ab77ce21c1e59
- tNMEA2000::InitCANFrameBuffers https://ttlappalainen.github.io/NMEA2000/classt_n_m_e_a2000.html#ac7dbb8a8e30a3926b5a2076e441d80eb

If you are not using Arduino environment, you have to do next:
- implement uint32_t millis(); if that does not exist on your environment. This has been declared as extern on N2kTimer.cpp line 46
- inherit N2kStream and write required methods. See N2kStream.h/.cpp 

Read document carefully and specially "Using tNMEA2000" https://ttlappalainen.github.io/NMEA2000/pg_lib_ref.html#autotoc_md35

phylib, 2024-02-08:
Thanks for your detailed answer. I'm going to try as you suggested and let you know about the state of my implementation.

ttlappalainen, 2024-02-08:
I forgot to mention one very important thing about CAN driver. You may have noticed that I have copy of CAN libraries under my git and some CAN libraries like NMEA2000_Teensyx and NMEA2000_esp32 has build in copy of original source. Reason for that is NMEA2000 fastpacket, which does not exist on CAN. Fastpacket consists of multiple my CAN frames, which has to be sent on sequential order. E.g., Teesies chips (also NXP) has several mailboxes, which has different priorities. If one simply sends next frame by using next free mailbox as normal CAN drivers does, it will brake fastpacket frame order and so those messages will be lost. If you look NMEA2000_Teensyx driver, it uses always same mailbox for same priority frame and so each fastpackect will be sent by using same frame. In this way fastpacket frames will be sent in order. This solution also forces higher priority frames to be sent before lower priority in rare concurrent situations - this I have prooved with my NMEA Simulator.

Other reason for including original driver source to NMEA2000 library driver source was that I get at beginning very frustrated that driver writers did not care of compatibility. They changed interface without keeping previous, which leaded sityation that I had to update my drivers all the time. Also like with NMEA2000_teensy they refuced to took my succestions fro making them NMEA2000 compatible. NMEA2000_teensy does not have FlexCAN library build in, but it requires FlexCAN library from my git and may now work with FlexCAN provided with Teensyduino. This is also very confusing for beginners.

Since you are going to use NXP, check first if it is compatible with NMEA2000_Teensyx which is actually for NXP i.MX RT1060 and works with defines with earlier Teensies also using NXP MCU.

phylib, 2024-02-08:
Now it gets little more complicated 🙈  I'll take a look at the CAN drivers and the fastpacket concept and hope that the Teensyx driver works. I hope I'll get that up and running.

You mentioned your NMEA simulator. I have not found information about that. Can you give me a pointer? A computer-based simulator software for testing if the implementation on my controller works correctly would be really helpful.

ttlappalainen, 2024-02-08:
You can find NMEA Simulator from https://kave.fi/Apps/  Full commercial version gives also good message analyzing features.