ttlappalainen/NMEA2000 #387: Sending more than 18 satellites with a 129540

titio72, 2024-02-16:
Hi,

My gps receiver tracks up to 64 satellites but adding them all to a message would breach the message size limit.
I guess I must use multiple message: what is the right way to do it?

Thanks,

ttlappalainen, 2024-02-17:
Read https://github.com/ttlappalainen/NMEA2000/issues/284#issuecomment-1940451526 and the issue conversation. There is ready code for reading and sending satellites.

As I described in issue (https://github.com/ttlappalainen/NMEA2000/issues/284#issuecomment-1937054142) you can send more by using TP message, if you modify limits on library. But then you have problem that TP message sending is slow except if you use addressed sending (https://github.com/ttlappalainen/NMEA2000/issues/284#issuecomment-1940467731).

ttlappalainen, 2024-02-17:
Forgot to mention that behaviour also depends of receiving device. On issue 284 user had MFD, which swapped satellites, if different lists were sent with 129540. Then it may be that device does not accept long TP message.

ttlappalainen, 2024-02-17:
Note that code has filter to limit satellites
```
static double SNRSendLimit=2;
static double ElevationSendLimit=DegToRad(5);
```