ttlappalainen/NMEA2000 #388: Padding Character for AIS static Data in PGN 129809

lsoltero, 2024-02-19:
The current code pads vessel names that are shorter than 20 characters with 0xFF.  I believe this is correct...

void SetN2kPGN129809(tN2kMsg &N2kMsg, uint8_t MessageID, tN2kAISRepeat Repeat, uint32_t UserID, char *Name, tN2kAISTranceiverInfo AISinfo)
{
 ...
    N2kMsg.AddStr(Name, 20);

and

void tN2kMsg::AddStr(const char *str, int len, bool UsePgm) {
  SetBufStr(str,len,DataLen,Data,UsePgm,0xff);
}

which defines 0xff as the padding...

the reason I ask...is that on RayMarine Axiom displays we are seeing the attached
[image]

so.. for example "CARMANA WIND" shows ups a "CARMANA WINDXXXXXXXX" suggesting that for Raymarine (anyway) the padding character is something other than 0xFF.

Thoughts appreciated...

--Luis

ttlappalainen, 2024-02-20:
AIS name requires as fillchar @ instead of 0xff. Change tN2kMsg as
```
  void AddStr(const char *str, int len, bool UsePgm=false, unsigned char fillChar=0xff);

//*****************************************************************************
void tN2kMsg::AddStr(const char *str, int len, bool UsePgm, unsigned char fillChar) {
  SetBufStr(str,len,DataLen,Data,UsePgm,fillChar);
}

```
and modify 
```
void SetN2kPGN129809(tN2kMsg &N2kMsg, uint8_t MessageID, tN2kAISRepeat Repeat, uint32_t UserID, char *Name) {
    N2kMsg.SetPGN(129809L);
    N2kMsg.Priority=6;
    N2kMsg.AddByte((Repeat & 0x03)<<6 | (MessageID & 0x3f));
    N2kMsg.Add4ByteUInt(UserID);
    N2kMsg.AddStr(Name, 20,false,'@');
}

```
Parser should be changed to
```
bool ParseN2kPGN129809(const tN2kMsg &N2kMsg, uint8_t &MessageID, tN2kAISRepeat &Repeat, uint32_t &UserID, char *Name, size_t NameBufSize) {
    if (N2kMsg.PGN!=129809L) return false;

    int Index=0;
    unsigned char vb;

    vb=N2kMsg.GetByte(Index); MessageID=(vb & 0x3f); Repeat=(tN2kAISRepeat)(vb>>6 & 0x03);
    UserID=N2kMsg.Get4ByteUInt(Index);
    N2kMsg.GetStr(NameBufSize,Name, 20, '@', Index);

    return true;
}
```

Parser interface should be change without using new parameter default value, which causes error in compilation for old users, but forces them to provide buffer length for safety.

Same problem is also with PGN 129810 Vendor ID and Call sign.

Test code and inform, so I'll add this to next update.

lsoltero, 2024-02-20:
It looks like 129794 needs this as well for Name, Call Sign, and Destination.

I will make the mods, test, and revert shortly.

Thanks for looking into this.

--Luis

ttlappalainen, 2024-02-20:
No. According document those has 0xff padding.

ttlappalainen, 2024-02-20:
Actually may be anyway. Other PGNs has special mention that they should be padded with @, but this does not. Anyway all has same data format. And AIS had that different behaviour - again one stupid thing in definitions.

lsoltero, 2024-02-20:
oh my... I would think that AIS would want standard ascii null terminators for all strings.  Can't see why 129794 would be different than others.  So bug in the documentation?  I think I will add these with '@' unless you are sure that these should be padded with 0xff.

ttlappalainen, 2024-02-20:
There is not mention on document. The only way is to test behaviour for 129794 and 129040. PGN 129809 and 129810 has clear definition that padding is @. I have not checked does canboat project have tested. Unfortunately that is neither proof. And neither are big manufacturers - they also have bugs.

lsoltero, 2024-02-20:
I just got confirmation that Class A targets on Raymarine display exhibit the same string padding/display issues as Class B targets... So I suspect that (for ray marine anyway) @ is required for Class A.  I will patch 129794 and 129040(although we don't use 129040), let me testers test, and revert back...

but right now I suspect that the documentation is not complete and omitted the line termination comment unintentionally.

lsoltero, 2024-02-21:
the attached patch fixes the padding issue for AIS.  Note that 
1. we are running a forked version of the repo so the patch may not apply directly to head
2. we don't use 129040 (and not present in our version of N2kMessages.cpp) so its not included.
3. 129794 is patched and class A AIS targets display correctly. So the documents must be incomplete.  

[0008-fixup-AIS-strPadding.patch](https://github.com/ttlappalainen/NMEA2000/files/14363021/0008-fixup-AIS-strPadding.patch)

lsoltero, 2024-02-21:
Here is a photo of a Class A target generated by the library

[image]

ttlappalainen, 2024-02-22:
Thanks. Also thanks for testing. I wonder was person who originally added function ever tested it. I have not been always strict and asked was function tested or not - I know that some functions has been added without proper testing.

Could you do next:
- Add to N2kMessages.h modified parsers parameter list comments too. E.g., * \param CallsignBufSize  Size of the Callsign buffer
- Add to N2kMsg.h modified AddStr parameter list comments
- Add to Documents/src/changes.md at top new update. Check format from earlier updates.
First note something like - *Compatibility change:* Parsers ParseN2kPGN129810 and ParseN2kPGN129794 parameter list changed. See below.
- On library.properties and library.json change version to 4.21.1
- Make PR

lsoltero, 2024-02-22:
Its possible that they did test...  My Garmin users never saw a problem. I suspect that Garmin and possibly others display only printable text and ignored non-printable characters.  Obviously not the case with Raymarine.

I will make the appropriate changes and send you a pull request.

ttlappalainen, 2024-02-24:
In library document tehre is for AIS strings e.g., "Call Sign - 7x -> 6 bit ASCII characters, ". Have you tested will it accept also lower letters like Vessel instead of VESSEL? N2k document does not limit that.

That hint and interface is actually wrong. Function should accept full ascii and do conversion, if lower letters can not be used. Also "6 bit ASCII" is not exact, since there are several variants.

If you like and have time:
- test lower letters
- if they does not work add temporary local buffer and do conversion so that interface accepts also lower letters
- fix AIS string comments. E.g., Call sign, max 7 chars. Call sign will be converted to capital. (last in case lower does not work.
- increment version -> 4.21.2 (meaning that this is just small fix)
- make PR

ttlappalainen, 2024-02-24:
https://en.wikipedia.org/wiki/Six-bit_character_code has special mention "SixBit ASCII (used by [AIS])". So definetely hint for strings is wrong.

Note that version has to be changed to both library.properties and library.json. I updated that and also added comment for changes.md.

lsoltero, 2024-02-24:
Hi Temo,

yes.. the character strings in AIS need to be in UPPER case letters.  In our code we make sure we pass UC strings to library.  But I see that you are suggesting that the library should actively convert the string to upper case.  Probably some additional filtering would also be needed to remove any symbols that are not supported in the encoding.  

The AIS methods could definitely be changes to automatically (and quietly) do the conversion... I noticed that the type being passed for these strings is (char *) and not (const char *) So... the routines could modify the strings without allocating a buffer.   

If the code were to allocated a buffer todo the character encoding then it would make sense to set the type for the arguments to (const char *)

Thoughts?

ttlappalainen, 2024-02-24:
Document does not require upper case for NMEA2000 but refers ITU-R M.1371-1

Parameters should be const char * - the thing I have not checked carefully when accepting PR. For function, which is not meant to modify parameters, they should be provided as const! It is never good to modify parameters in this kind of function - never do that.

It does not need to do dynamic allocation. It can use stack. Just use variable char CvtBuf[21];

lsoltero, 2024-02-24:
Hi Temo,

Agreed that the (char *) arguments should be changed to (const char *).   May I suggest that that approach be used for all library function that take (char *) arguments.  It seems that these are found only AIS related functions.

https://www.itu.int/dms_pubrec/itu-r/rec/m/R-REC-M.1371-1-200108-S!!PDF-E.pdf

specifically states

> Some telegrams specify the inclusion of character data, such as shipís name, destination, call sign, and more. These fields should use a 6-bit ASCII as defined in Table 14.

So.. although the N2K spec docs may not specifically prohibit lower case characters I believe the document (once again) is inconsistent.

My suggestion is to create a filter function that will build a text string by translating all lower case letters to upper case and then replace any symbol not in table 14 with '?'.  This should be done for all AIS functions that take (const char *) strings as arguments.

If you are OK with this I am happy to make the mods, test, and then issue a pull request.

LMK

Also should go check the AIS routines in the n183 parser.  They may have similar issues.

--Luis

ttlappalainen, 2024-02-25:
You did not mention did you actually test AIS strings in lower case letters. They have been capital since AIS protocol saves bits in telegrams specially in radio transport. Like in old days all computer programs used only capital ASCII, but now accept UTF also AIS may improve in future. It may be possible that some MFD:s accept also lower case. So did you test, what Raymarine results with lower case letters? I'll try to test it with Garmin.

If they do not accept lower case, then conversion is required. If some does, then conversion should be optional. Optional parameter can be added later, if necessary.

I mentioned hint "6 bit ASCII characters" for AIS string is wrong. SixBit ASCII codes starts from 0 = @; A=1,..., which is not the case when you provide strings to functions. Neither strings on N2k bus will not be transfered as SixBit  ASCII. Proper hint would be something like "String will be converted to acceptable AIS string. Only characters within SixBit ASCII character set will be accepted." Proper hint depends of behaviour, which should be tested.

Conversion could be:
size_t StringToAISString(const char *str, char *AISStr, size_t AISStrBufSize) {
...
}
return value is number of characters copied to AISStr

lsoltero, 2024-02-25:
Hi Temo,

Whether MFDs accept and display lower case letters received in AIS N2K PGNs will be specific to the manufacturer.  We can test some of the usual suspects (Raymarine, Navico, Garmin) but it will be impossible to do them all.    Case in point.. When I reported the 0xff string padding issue... Garmin and Navico displayed the strings correctly but Raymarine did not.  It seems that Garmin and Navico filter out non printable characters where as Raymarine does not.  So some manufactures might handle lower case letters but others might not. 

It's unfortunate that NMEA was not more specific about the character encoding for these PGNs.  I don't have access to the document so can't know exactly what is stated.  However,  I do know the following.

AIS traffic originates for VHF transmissions using a very strict standard referenced in ITU-R M.1371-1.  This standard specifies that all letters are in upper case and that not all symbols are supported ( {} for example ).  So data that is placed on the N2K bus by AIS receivers by default will be transcoded into the equivalent ASCII characters defined in table 14. And we know for that all MFDs should support ASCII versions of all characters defined in that table.

What we don't know (unless every single manufacturers equipment is tested) is whether characters/symbols outside of table 14 can be displayed and what they might look like on the screen.

It is not possible for us to test ALL N2K/AIS capable displays.

Here is what I do know.
1. All AIS receivers support table 14
2. All AIS/N2K Displays support the ASCII equivalent of elements in table 14
3. We have access to the entire universe of AIS targets received via satellite, mobile AIS stations, fixed AIS stations (over 38 million targets last I checked) and every string entry that I have seen in this database conforms to the ASCII equivalent of symbols in table 14.
4. Our products (as of this moment) are reporting over 5000 AIS targets per minute around the globe.  And strings in those targets all conform to table 14.

So.. given the above, when creating AIS targets that do not originate from over the air VHF transmissions, we make sure that strings comply with the ASCII characters defined in table 14.  We do this before we make calls to the library and so have not run into issues.

My recommendation would be one of the following...

1. Leave things as is but warn users that there might be display issues unless strict adherence to table 14 is followed.

or

2. Have the library convert input strings into ASCII 6 bit characters by converting a-z to A-Z and replacing symbols not in the table (i.e. {,},|, etc...) with ?.

So to answer you specific question...  No I have not tested lower case letters on MFD and not inclined to do so.

Anyway, I am happy to do the implementation for (2) if you like although our code does not currently require this.

LMK.

--Luis

ttlappalainen, 2024-02-25:
As I said it may change in future... Anyway it is easy to add later parameter to pass conversion if necessary. So please do.

But once again. Library function parameter is 0 terminated ASCII string not SixBit ASCII! Conversion should then convert all characters 0x01-0x1f -> '?' and >0x60 -> '?' . I do not know what should do if string contains '@', which is termination or NA character.

lsoltero, 2024-02-25:
ok. will do... 

I think if @ is in the middle of a string then we should just leave as is and continue until we see '\0' or i == AISStrBufSize no harm is done and the N2K display will stop the string when it see @ with anything trailing being ignored.

lsoltero, 2024-02-26:
Modifications completed.

code tested in Raymarine MFD.  

Created pull request.

lsoltero, 2024-03-19:
Hi Timo,

I finally spent the week merging HEAD with my 3 year old fork.. and discovered that.  The version of AddAISStr() that was checked in does not work.  I had been using the version that had the extra buffer that you wanted to eliminate.  The following code has been tested and works.

Did you want me to do a specific PR for this?  I have a BIG patch coming your way with changes to the AIS code in the library.. I can incorporate this into that PR when ready.. currently testing all those changes.

LMK

```
diff --git a/src/N2kMsg.cpp b/src/N2kMsg.cpp
index 090f1ee..f3387a5 100644
--- a/src/N2kMsg.cpp
+++ b/src/N2kMsg.cpp
@@ -220,16 +220,17 @@ void tN2kMsg::AddStr(const char *str, int len, bool UsePgm, unsigned char fillCh
 // https://www.itu.int/dms_pubrec/itu-r/rec/m/R-REC-M.1371-1-200108-S!!PDF-E.pdf
 void tN2kMsg::AddAISStr(const char *str, int len) {
   unsigned char *buf=Data+DataLen;
-  for (; len>0 && *str!=0 && DataLen<MaxDataLen; len--, DataLen++, buf++, str++) {
-    char c = toupper((int)*str);
-    *buf=(c >= 0x20 && c <= 0x5F) ? c : '?';
+  int i=0;
+  for (; i <len && *str != '\0' && DataLen<MaxDataLen; i++, DataLen++) {
+    char c = toupper((int)*str++);
+    *buf++=(c >= 0x20 && c <= 0x5F) ? c : '?';
   }
-
   if ( len > MaxDataLen-DataLen ) len=MaxDataLen-DataLen;
-  if ( len>0 ) memset(buf,'@',len);
+  for (; i<len; i++, DataLen++ ) {
+    *buf++ = '@';
+  }
 }
 
-
 //*****************************************************************************
 void tN2kMsg::AddVarStr(const char *str, bool UsePgm) {
   int len=(str!=0?strlen(str):0);

```

ttlappalainen, 2024-03-20:
I do not see quicly difference.

I also prefer incrementing str and buffer on loop. At least I feel it more readable.
memset should be also better than for loop. Just need to have Datalen+=len;

lsoltero, 2024-03-20:
Hi Timo,

The code above has been tested and works.  Given the small buffer sizes the efficiency gains using memset will be minor given that you have to push/pop arguments onto a call stack to make the function call, etc...

making the minor code changes you suggest would require another round of testing which is a big deal since we don't have ray marine MFD's here to test with.  We would need to create a special build and send out to testers in the field and have them report back.  Not worth the effort in MHO.

So.. I will prepare a PR for the above patch if that is OK with you.

I do have a big patch comming 

    Add PGN 127510 and 127750
    Modify PGN 129038, 129039, 129794, 129809, and 129810 to include AIS Transceiver Info and SID (where applicable)

That has been tested...

would you prefer for me to submit one large PR with the above fix or two PRs one for the AddAISStr() and a second for the mods to other AIS and new PGNs.

--Luis

ttlappalainen, 2024-03-25:
I fixed and tested SetAISStr. Use published version.

You did not either tested changed code. ParseN2kPGN129810 did not compile, since declaration was wrong.

lsoltero, 2024-03-25:
Hi Timo,

Thanks for the SetAISStr fix.. I will back out my changes and merge my code with your version.

Interesting about 129810... We use this PGN extensively.  I am trying to understand why my PR might have been different than the code that we what we use.  

I will merge your changes with my updated version of all the AIS PGNs and submit a PR for those. Once our libraries are in sync with yours we should have much fewer problems moving forward.

lsoltero, 2024-03-25:
your version of SetAISStr() has now been tested and it works as expect.