ttlappalainen/NMEA2000 #395: Linking Error: undefined reference to `ParseN2kPGN129810(

ronzeiller, 2024-03-24:
May be someone can help.
Updated my N2kToNMEA0183 AIS library to the newest NMEA2000 Library.
Everything is fine but get an error while Linking:

`undefined reference to `ParseN2kPGN129810(tN2kMsg const&, unsigned char&, tN2kAISRepeat&, unsigned int&, unsigned char&, char const*, unsigned int, char const*, unsigned int, double&, double&, double&, double&, unsigned int&)'`

Very strange because all other 
ParseN2kPGN129038
ParseN2kPGN129794
ParseN2kPGN129039
ParseN2kPGN129809

are working, compiling and linking fine. 
(if I disable ParseN2kPGN129810)

Code is:

`void N2kToNMEATask::HandleAISClassBMessage24B(const tN2kMsg &N2kMsg) {

  uint8_t _MessageID;
  tN2kAISRepeat _Repeat;
  uint32_t _UserID, _MothershipID;
  char _Vendor[4];
  size_t _VendorBufSize;
  char _Callsign[8];
  size_t _CallsignBufSize;
  uint8_t _VesselType;
  double _Length;
  double _Beam;
  double _PosRefStbd;
  double _PosRefBow;

  if ( ParseN2kPGN129810(N2kMsg, _MessageID, _Repeat, _UserID,
                        _VesselType, _Vendor, _VendorBufSize, _Callsign, _CallsignBufSize, _Length, _Beam,
                        _PosRefStbd, _PosRefBow, _MothershipID) ) {`

Error Message is:
Linking .pio/build/esp32dev/firmware.elf
undefined reference to `ParseN2kPGN129810(tN2kMsg const&, unsigned char&, tN2kAISRepeat&, unsigned int&, unsigned char&, char const*, unsigned int, char const*, unsigned int, double&, double&, double&, double&, unsigned int&)'

Same if I compile in Arduino:
lib/NMEA0183-AIS/Examples/NMEA2000ToWiFiAsNMEA0183WithAIS/N2kDataToNMEA0183.cpp:495: undefined reference to `ParseN2kPGN129810(tN2kMsg const&, unsigned char&, tN2kAISRepeat&, unsigned int&, unsigned char&, char const*, unsigned int, char const*, unsigned int, double&, double&, double&, double&, unsigned int&)'

How can it be that all Parsers are found but not the 129810?

Thanks for any ideas

ttlappalainen, 2024-03-25:
I published a fix. Please update library.

ronzeiller, 2024-03-25:
Thank you Timo, everything is working fine now!

closed 2024-03-25 by ronzeiller