ttlappalainen/NMEA2000 #398: Parsing of AC PGNs (127744 - 127749

mar0997, 2024-04-17:
Hi Timo - 

Really appreciate the work that you have done with the library.  I am trying to make an update to my system to pull in AC parameters.  The most important ones for me are current and power.  From looking at some previous issues that were resolved along with the library I put together what is shown below for PGN 127745.  The syntax is correct as it compiles, but it is not pulling in the data.  I have been looking at it for a couple of days and still can't figure out what I am doing wrong.  Can you help point me in the right direction?

 bool ParseN2kPGN127745(const tN2kMsg &N2kMsg,unsigned char &SID,double &ConnectionNumber, double &ACRMScurrent,double &Power) {
          if (N2kMsg.PGN!=127745L) return false;
          int Index=0;
          SID=N2kMsg.GetByte(Index);
          ConnectionNumber=N2kMsg.Get2ByteUDouble(1,Index);
          ACRMScurrent=N2kMsg.Get2ByteUDouble(0.1,Index);
          Power=N2kMsg.Get2ByteUDouble(0.1,Index);
          return true;
 }

ttlappalainen, 2024-04-17:
- Do you ever call the parser? See example DataDisplay
- Do you have message on the bus?

mar0997, 2024-04-17:
I do call the parser and am successfully parsing other messages.  Here is where the parser gets pulled in:

void loop() {
  SendN2kRudder();
  SendN2kTemperature();
  SendN2FluidLevel();
  SendN2kPGN128777();
  NMEA2000.ParseMessages();
}

I did notice with these messages that I am not able to include them in the message handler.  I believe that is because I am doing everything in my local file rather than referring off to the h and .cpp file.  But other than that everything else is the same as how I am parsing other messages.

The message I am attempting to parse is on the bus as shown below (the other phases are as well).  And I can also see the message I am sending is on the bus, but with no AC current/power data (I'm only trying current/power for now).

[image]

mar0997, 2024-04-17:
My mistake - I accidentally closed the issue rather than submitting the above reply.

ttlappalainen, 2024-04-18:
But do you call your parser on your message handler? There is no automatic call for each message parser and also message handler has to be registered.

Parsers nad setters can be locals, if they are not published on N2kMessages module.

mar0997, 2024-04-19:
Thanks for the input.  I believe I understand the comment and I made a couple of changes and added "ACPhaseB" to message handler and to the message.  I can't get it to successfully compile when I added "{127745L,&ACPhaseB}," to the message handler so I know I still have something that isn't correct. 

//Message handler for received message parameters
void Rudder(const tN2kMsg &N2kMsg);
void OutsideEnvironmentalParameters(const tN2kMsg &N2kMsg);
void FluidLevel(const tN2kMsg &N2kMsg);
void ACPhaseB(const tN2kMsg &N2kMsg);

tNMEA2000Handler NMEA2000Handlers[]={
  {127245L,&Rudder},
  {130310L,&OutsideEnvironmentalParameters},
  {127505L,&FluidLevel},
  {127745L,&ACPhaseB},
  {0,0}
};

 bool ParseN2kACPhaseBPGN127745(const tN2kMsg &N2kMsg,unsigned char &SID,double &ConnectionNumber, double &ACRMScurrent,double &Power) {
          if (N2kMsg.PGN!=127745L) return false;
          int Index=0;
          SID=N2kMsg.GetByte(Index);
          ConnectionNumber=N2kMsg.Get2ByteUDouble(1,Index);
          ACRMScurrent=N2kMsg.Get2ByteUDouble(0.1,Index);
          Power=N2kMsg.Get2ByteUDouble(0.1,Index);
          return true;
 }

ttlappalainen, 2024-04-19:
You have only declared parser, but still do not call it. You need to also write full function `void ACPhaseB(const tN2kMsg &N2kMsg);`. See example of other functions  like `void Rudder(const tN2kMsg &N2kMsg);` how they handle message, call related parser and handle parsed data.

mar0997, 2024-04-22:
I decided to create my own .h and .cpp files and successfully integrated the AC current/power, voltage/frequency and AC input status pgns into my file.  I really appreciate all of the guidance that you provided.

closed 2024-11-21 by mar0997