ttlappalainen/NMEA2000 #400: (help)im stock in 8bit

gravefunc, 2024-04-28:
I'm working on an STM32-based application that involves processing NMEA 2000 CAN messages, and I'm running into a problem with handling data payloads larger than 8 bytes.

Specifically, I'm working with the Engine Dynamic Parameters PGN (127489), which has a data payload of 12 bytes. However, in my current implementation, I'm only able to read the first 8 bytes of the data payload using the HAL_CAN_GetRxMessage() function provided by the STM32 HAL library which is used https://github.com/mehrdad987/stm32-NMEA2000/blob/main/main.c.

I've included the relevant code snippet below for your reference:

` void HAL_CAN_RxFifo0MsgPendingCallback(CAN_HandleTypeDef *hcan) {
    CAN_RxHeaderTypeDef rxHeader;
    uint8_t rxData[16];

    if (HAL_CAN_GetRxMessage(hcan, CAN_RX_FIFO0, &rxHeader, rxData) == HAL_OK) {
        // Extracting priority, PGN, source address, and device instance from the CAN ID
        uint8_t priority = (rxHeader.ExtId >> 26) & 0x07;
        uint32_t pgn = (rxHeader.ExtId >> 8) & 0x3FFFF;
        uint8_t sourceAddress = rxHeader.ExtId & 0xFF;
        uint8_t deviceInstance = (rxHeader.ExtId >> 18) & 0x1F;
        uint8_t dataLength = rxHeader.DLC; `
        
I was hoping you could provide some guidance or suggestions on how I can modify the code to successfully retrieve the complete data payload, even if it exceeds the standard 8-byte CAN frame size.

ttlappalainen, 2024-04-28:
Do you actually use the library or not? Library takes case of reading data properly provided as fastpacket or TP. You just need STM driver for library.

gravefunc, 2024-04-28:
NO I didn't use the library cause couldn't find examples that worked with stmcubeide
where can I find reading data process

ttlappalainen, 2024-04-28:
You can try to ask from https://github.com/thomasonw/NMEA2000_socketCAN He did some work with STM32 and maybe has driver for my library. If you decide not to use my library, you are free to read what it does.

Why not to use something which is already supported and tested like ESP32 or Teensy 4.x?

gravefunc, 2024-04-29:
Actually, I have used your library with an ESP32, and it is working well in my boat. Thanks to you! Now, I also have some STM32F103, and I’m interested in finding a way to use them. 
However, the edited code for STM32 https://github.com/mehrdad987/stm32-NMEA2000is working the raw data is correct but it seems to have issues with data length which is all the time shows 8

jiauka, 2024-04-29:
I made  it work some time ago. it is a WIP,but could help you
https://github.com/jiauka/NMEA2000_stm32f1