ttlappalainen/NMEA2000 #402: Not an issue, but a support request or feature request

Leijoma, 2024-06-10:
I want to forward the raw nmea2000 message to the serial stream. Is that possible, and if so, how do i do that? I wan to read the data with the CanView software.

ttlappalainen, 2024-06-10:
Not directly. To do that you should do same as inside library and split message to CAN ID and data frames. In case of message is fast packet type (even data length<=8), you must build separate fast packet frames.

Why not buy a license and use my NMEA Simulator or even buy the whole box.

Leijoma, 2024-06-11:
Ok
What i want to achieve is to analyze what is being transferred on the N2K
network. I guess i can use the NMEA simulator and just send the data in
actisense format to the simulator from my ESP32. No need to send the raw
data.

ttlappalainen, 2024-06-11:
Simulator does both simulation and bus data analyzing, but analyzing requires license.

Leijoma, 2024-06-11:
Aha. I get it. Thanks.

svdrummer, 2024-06-11:
> Simulator does both simulation and bus data analyzing, but analyzing requires license.

Is there a link to simulator.. i cant find it.

ttlappalainen, 2024-06-11:
https://www.kave.fi/Apps/