ttlappalainen/NMEA2000 #406: Adding NMEA0183 Sentences, and some other questions

amjpike, 2024-06-16:
I have been working on a solution based on the ESP32, using your code. The solution receives a N2K Source and translates to NMEA0183 through a wired connection, and also transmits the NMEA0183 over Wifi. There are several things though that I have been trying to add,

1) How to add support for the following sentences, RMB, VLW, VWR, GLL, HDM, MTW, I don't understand the routines enough to add them, so any additional documentation would be helpful.
2) I want to have an 115K output and a 4,800 output. However when I enable the 4,800 the rapid update N2K messages are converted but it never updates the slower ones such as DPT, or if it does every few minutes
3) My ultimate solution is to send NMEA0183, 4800 through one serial port, Activsense Protocol through a USB connection and then a NMEA0183 Wifi Server for Ipads, etc. so if anyone has done this please let me know.

ttlappalainen, 2024-06-17:
Check https://docs.hatlabs.fi/sh-wg/ and https://github.com/hatlabs/SH-wg-firmware

amjpike, 2024-06-20:
I went through that, but I cannot see where I can add support for those sentences.

ttlappalainen, 2024-06-23:
Have you checked https://github.com/ttlappalainen/NMEA0183? It already has support for at least GLL, HDM. You can use those as example and add own routines to your own code.

The problem with your 4800 may be on data sending. You should never use blocking data reading or writing or actually any blocking methods in you main loop. With slow serial you easily reach blocking state, when your serial buffer gets full.