ttlappalainen/NMEA2000 #408: CAN Transceiver MCP2562 8mb/s

greenhouse69, 2024-06-28:
I am having problems with my new 8mb/s MCP2562 (MICROCHIP MCP2562FD-E/MF). The problem is that when I connect power from nmea2000 the board does not boot and it is necessary to press the reset of the esp32 to get it to start working.  If I connect directly to the usb of the esp32 it works correctly.  Originally I used a 1 mb/s MCP2562 and everything worked fine. My question is whether the 8mb/s MCP2562s produce errors or if it could be something else. 

Thanks

ttlappalainen, 2024-06-29:
I have noticed some esp32 versions getting stuck on error. I have auto reset feature for library, but I have not published it.

jiauka, 2024-06-29:
Do you have the schematic? Could be that CAN bus transceiver is connected to some bootstrap pins.

greenhouse69, 2024-06-29:
> I have noticed some esp32 versions getting stuck on error. I have auto reset feature for library, but I have not published it.

Timo, yes I have been studying the firmware of the esp32 in its different versions (the most modern ones have some boot problems). I have updated the firmware with esptool, and I have tried several esp32 (some new ones and some old ones that worked) but the problem remains.
The automatic reboot option is very interesting for these cases. If you can send me a code example to help me to understand the automatic restart it would be much appreciated.
In my case when I connect it from the mains from N2k it blocks it completely, it doesn't start anything (neither the wifi) until I press the reset, then it works perfect.
Thanks for your help.

Translated with DeepL.com (free version)

greenhouse69, 2024-06-29:
> jiauka

Jiauka, this is the schematic:
[image]

greenhouse69, 2024-06-29:
Has anyone tried the MCP2562 version H, speed 8mb/s?

ttlappalainen, 2024-06-30:
I actually meant that CAN controller gets stuck on bus error and needs to reset. If your WiFi does not work, then it sounds strange.

Is your system isolated or should you actually use ISO1050?

svdrummer, 2024-06-30:
Try another can controller.

ttlappalainen, 2024-06-30:
One can not change CAN controller on ESP32 except by using external one. Did you mean "CAN transceiver"?

svdrummer, 2024-06-30:
Apologies, I meant CAN controller. Isolated units are the only way to go, if you want to avoid stay current loops.  I have seen quite a few laptops blow thier USB ports because they dont use idolated USB interfaces. I just play a numbers game, if I am the only one having an issue, then the issue must be related to my hardware.

ttlappalainen, 2024-06-30:
?? CAN controller ?? You make isolation by using isolated CAN transceiver. You do not need to change CAN controller!

If one has just N2k->WiFi converter or just e.g., optoisolated digital inputs, one does not need isolation. Anyway as you said it is risky to have device unisolated device connected to USB and N2k bus at same time.

jiauka, 2024-06-30:
But the MCP2562FD is not isolated... Could you share full schematic?

greenhouse69, 2024-06-30:
My project does not require an isolated tranciever. I only use the usb for test and control of the esp32. The system gets the current from the N2k network (12v) which is converted with a 5v current regulator, type Lm7805, which is connected to the esp32. Sensors do not need to be isolated, gnd is taken from the esp32 board.

greenhouse69, 2024-06-30:
My project works correctly with a 1mb/s MCP2562, I upgraded the board to get different improvements like touch switches, control led and upgraded to a smaller 8mb/s MCP2562 in DFN format with 8 pins + Thermal pad. I think I have connected something wrong on the thermal pad that makes it crash the system.