ttlappalainen/NMEA2000 #411: Send Failed Question

atestani, 2024-07-08:
I am working on a custom ESP32-wroom-E32 board and am getting Send Failed errors after a delay depending on the size of the buffer I create.  At the moment I am using the Battery Monitor example with everything as default except my CAN pins which are declared before the includes. I am sure I have TXD/RXD for the ISO1050  transceiver I am using and that the bus is terminated properly and CANH and CANL are connected properly.   Everything looks OK electrically in that I can see  what look like good CANL and CANH signals on an oscilloscope.   At this point I am out of ideas for why my messages are not being sent.

I understand I need to have another device on the bus or the buffer will fill up due to unsent messages.  Question: Is having an Actisense NGT1 enough as a device on the bus in addition to my own?  The NGT1 is not showing any activity from my device.

atestani, 2024-07-08:
Some additional information:
Includes look like this:
```
#include <Arduino.h>
#define ESP32_CAN_TX_PIN GPIO_NUM_5
#define ESP32_CAN_RX_PIN GPIO_NUM_4
#include "NMEA2000_CAN.h"       // This will automatically choose right CAN library and create suitable NMEA2000 object
#include "N2kMessages.h"
```
I added this to see if the NMEA2000.Open() worked:
```
  Serial.print("NMEA2000.Open returned: ");
  Serial.println(NMEA2000.Open());
```
and the result is: _NMEA2000.Open returned: CAN device ready_

Here is the start of the serial monitor trace:
```
Start address claim for device 0
1553 : Pri:6 PGN:60928 Source:25 Dest:255 Len:8 Data:1,0,C0,FF,0,AA,46,C0
2054 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
2064 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
2074 : Pri:6 PGN:127513 Source:25 Dest:255 Len:8 Data:1,D1,1,A4,1,35,7D,4B
3554 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
3564 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
5054 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
5064 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
6554 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
6564 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
7074 : Pri:6 PGN:127513 Source:25 Dest:255 Len:8 Data:1,D1,1,A4,1,35,7D,4B
8054 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
8064 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
9554 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
9564 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
11054 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
11064 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
11554 : Pri:7 PGN:126993 Source:25 Dest:255 Len:8 Data:60,EA,0,FF,FF,FF,FF,FF
12074 : Pri:6 PGN:127513 Source:25 Dest:255 Len:8 Data:1,D1,1,A4,1,35,7D,4B
12554 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
```
and when it starts failing it looks like this:
```
84564 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
PGN 127508 send failed
86054 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
PGN 127506, frame:0/2 send failed
86064 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
PGN 127513, frame:0/2 send failed
87074 : Pri:6 PGN:127513 Source:25 Dest:255 Len:8 Data:1,D1,1,A4,1,35,7D,4B
PGN 127508 send failed
87554 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
PGN 127506, frame:0/2 send failed
87564 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
PGN 127508 send failed
89054 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
PGN 127506, frame:0/2 send failed
89064 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF

```

atestani, 2024-07-08:
FWIW, Here is a scope trace:
[image]

ttlappalainen, 2024-07-08:
You need to triple check:
- CANL/CANH order
- Bus termination, for short test at least one 120 ohm termination resistor. Prefered two.
- You must have at least one other device on the bus.
- Check that bus is powered or other devices powers completely from external source. E.g., NGT-1 and many MFD:s requires that bus is powered with 12V.

atestani, 2024-07-08:
Does the NGT-1 count as the other device on the bus?

ttlappalainen, 2024-07-08:
Yes.

atestani, 2024-07-08:
I am embarrassed to say how I fixed the problem.  When I went to check the bus power at 12V, I found that the NGT-1 was not connected to the bus!!!  When I connected it, it all works fine now.

Very sorry for the confusion!

closed 2024-07-08 by atestani